LSatan/SmartRC-CC1101-Driver-Lib #1: Incompatible with ESP8266 (Wemos d1 mini) board

bondar, 2018-03-28:
Hi,

Here are the error that I get when trying to compile it with ESP8266:

ELECHOUSE_CC1101_RCS_DRV.cpp:68:3: error: 'SPCR' was not declared in this scope
ELECHOUSE_CC1101_RCS_DRV.cpp:69:32: error: 'SPE' was not declared in this scope
ELECHOUSE_CC1101_RCS_DRV.cpp:69:43: error: 'MSTR' was not declared in this scope
ELECHOUSE_CC1101_RCS_DRV.cpp:70:9: error: 'SPSR' was not declared in this scope
ELECHOUSE_CC1101_RCS_DRV.cpp:71:9: error: 'SPDR' was not declared in this scope
ELECHOUSE_CC1101_RCS_DRV.cpp: In member function 'byte ELECHOUSE_CC1101::SpiTransfer(byte)':
ELECHOUSE_CC1101_RCS_DRV.cpp:82:3: error: 'SPDR' was not declared in this scope
ELECHOUSE_CC1101_RCS_DRV.cpp:83:12: error: 'SPSR' was not declared in this scope
\ELECHOUSE_CC1101_RCS_DRV.cpp:83:23: error: 'SPIF' was not declared in this scope

SPI is supported on that platform... thoughts?

LSatan, 2018-03-30:
Hello,

Thanks for your problem message.
The library is currently only optimized for Arduino Nano, Uno, Mega and others. Unfortunately, I can not currently offer support for esp8266 wemos d1 mini. Since I have no such module for tests.

stritti, 2018-06-30:
I have the same problem. Looking forward some help.

LSatan, 2018-07-26:
Hello
ESP is supported in version 1.4.
The update is coming this week.

LSatan

closed 2019-01-18 by LSatan