LSatan/SmartRC-CC1101-Driver-Lib #4: Sending Strings and reading (receiving) Intertechno commands

Falkenstone, 2018-07-25:
Hi,
first many thanks for the very useful lbrary. I've two points:

1. Could you implement a possibilty for sending and receiving strings (text) ? I know the possibility of sending character by character in a loop. But this seems very ineffective and not perfomant. 
2. How is it possible to read (receive) Intertechno commands from a remote Control. Sending commands is possible. In the example nothing happens by preesing a button from remote Control.

sorry i'm a newbie but i've no knowledge about rf.

best regards
Falke

LSatan, 2018-07-25:
Hello Falke

First of all, many thanks for your feedback.
So to point 1:
You can only send and receive what the RC switch library allows.
if you have another library, the text may send on the base of cheap RF modules like RC switch,
can you send me a link to this library?
Otherwise you should ask the question at rc-switch.
And of course you need to customize the example codes as you like.
Point 2:
I do not know if the rc-switch library can receive this kind of signal. You may need to adjust the receive bandwidth in the receive sketches.
from       //ELECHOUSE_cc1101.setRxBW(1);  // 1 = 58khz, 
to              ELECHOUSE_cc1101.setRxBW(16);  // 16 = 812khz.

regards
LSatan

Falkenstone, 2018-07-26:
Hi LSatan,
many thanks. I've realized first point with the RadioHead Library. It't not the favorite solution, because of 2 different libraries. I can't modifify the RC Switch library because of my missing knowledge where and what.
Second Point works too with by setting the bandwidth.
thx
Falke

LSatan, 2018-07-26:
hello hawk
Glad to hear that.
if I understood correctly, send with rc switch also.
So my problem is that I do not know what you intend to do. So what your code should do when and how

Falkenstone, 2021-06-17:
Hi Satan,

is it possible to use an CC1101 to receive text messages from another CC1101 sender and all other codes from a remote control like Intertechno? 
I saw a new method in your new SmartRC library. This works fine with text, but no other code from a remote control can be received. When i use the RC-Library all codes can be received but no text message. Is there a proper way to combine these methods?
My solution from the post above is that i use for text messages the RH_ASK Library and all other codes the RC-Switch Library in combination. But this works only for an cheap Modul as sender and not for a CC1101 modul and now i want to use an CC1101 for longer distance and text messages. 
Falke

LSatan, 2021-06-17:
Hello Falke, 
if you can send everything with a cheap module. then it works with the cc1101. your configuration may not be correct. Sketching from your transmitter and receiver can bring light into the dark.
Regards

Falkenstone, 2021-06-17:
Hi Satan, i took the example of sending and receiving "hello world" with CC1101 in the SmartRC Library. No problem, but the receiver can't receive any code from a remote control, only text from the sender. How can the CC1101 receive text messages from another CC1101 sender and the codes from any remote control. In the example for receiving "Hello...." no rc_switch is used and therefore no Intetechno code can be received. How can i solve it? I need a receiver which can receive text messages fom a CC1101 and the codes from a remote control like Intertechno. Single solution works but not in combination.
please help
thx

LSatan, 2021-06-27:
if i have some time i will take care of it.
Regards