LSatan/SmartRC-CC1101-Driver-Lib #10: Driver not transmitting with ESP (32 nor 8266)

mrbalintlorand, 2018-12-30:
Hi,

I tried setting up this awesome driver both on an ESP32 (Wemos Lolin32 with Oled) and and an ESP8266 (NodeMCU ESP-12E) to transmit, but I could never get signals from it. I tried the simplest code, the Send Demo included in the examples, and wired everything up, just as the documentation says.
To rule out other issues:
- Transmitting with an Arduino Nano works fine.
- Receiving works fine on the ESPs.

Did anyone try with ESPs?
Cheers

LSatan, 2018-12-31:
Hello,
that you can receive with ESP is already good. This means that Spi is properly connected and working.
Maybe you exchanged the Gdo Pins? Gdo2 has to go to D2 Gdo0 has to go to D1. The thing is that Gdo0 can be used not only to send but also to receive, but not vice versa so gdo2 to send. If this is all right, you can try changing the send pin. For ESP8266, set gdo0 to D1 and change the sketch (SendDemo_cc1101.ino).

from:
#ifdef ESP32
Pin = 2; // for esp32! Transfer to GPIO pin 2.
#elif ESP8266
Pin = 5; // for esp8266! Send to pin 5 = D1
#otherwise
Pin = 6; // for Arduino! Send on pin 6.
#endif

to:
#ifdef ESP32
Pin = 2; // for esp32! Transfer to GPIO pin 2.
#elif ESP8266
Pin = D1; // for esp8266! Send to pin 5 = D1
#otherwise
Pin = 6; // for Arduino! Send on pin 6.
#endif

So from 5 in D1 or in another available pin.
Also try to change the pin of Esp32.

But first test if GDO pins have not been confused.

It would be nice if you could give us some feedback as to whether and what worked for you.

Regards LSatan

mrbalintlorand, 2018-12-31:
Hi,

Thanks for the quick response! Yeah I checked the wiring multiple times, and pretty sure I followed the instructions properly. :( I used the same wires as when connecting to the Arduino, and that transmitted  just fine.
For the send demo, I tried it both with '5' and 'D1', but with no luck. (Btw I don't get it why is there on this platform a different D numbering than the GPIO numbers used in code... so confusing!)

I also tried adding this additional setup to the beginning of the setup function:
`  pin = D1;  // for esp8266! Transmit on pin 5 = D1
  ELECHOUSE_cc1101.setGDO(D1, D2); //Put the Gdo pins. For libraries that address the gdo pins directly.
  ELECHOUSE_cc1101.setSpiPin(D5, D6, D7, D8); //custom SPI pins. Set your own Spi Pins.`

Would there be a way to debug this more deeply? I really wonder what can be different on the ESPs that mess with transmitting.

Cheers,
Balint

LSatan, 2019-01-01:
Hi,
The gpio numbering is different from the D numbers is actually not good. But more indebted to the module. That one can also enter D numbers at the arduino ide is a noob addition.

ELECHOUSE_cc1101.setGDO (D1, D2); may not be used for rc-switch and newremoteswitch because otherwise the send does not work!

So there are many people who use the nodemcu with this library.
Without problems.

I looked a bit in the www. And seen that there are different pinouts for different nodemcu's.

For some modules, D1 is not a PWM output. But at D2 all have PWM. To receive no pwm is needed. It is needed to send.

So please try the following:

change the wiring.
Swap the gdo pins.
GDO0 to D2
GDO2 to D1

take the cc1101 default sketch as are (without addition).

in the send demo
from

#ifdef ESP32
pin = 2; // for esp32! Transmit on GPIO pin 2.
#elif ESP8266
pin = 5; // for esp8266! Transmit on pin 5 = D1
#else
pin = 6; // for Arduino! Transmit on pin 6.
#endif

to

#ifdef ESP32
pin = 2; // for esp32! Transmit on GPIO pin 2.
#elif ESP8266
pin = D2; // for esp8266! Transmit on pin 5 = D1
#else
pin = 6; // for Arduino! Transmit on pin 6.
#endif

in the receive demo
from

#ifdef ESP32
pin = 4; // for esp32! Receiver on GPIO pin 4.
#elif ESP8266
pin = 4; // for esp8266! Receiver on pin 4 = D2.
#else
pin = 0; // for Arduino! Receiver on interrupt 0 => that is pin # 2
#endif

to

#ifdef ESP32
pin = 4; // for esp32! Receiver on GPIO pin 4.
#elif ESP8266
pin = D1; // for esp8266! Receiver on pin 4 = D2.
#else
pin = 0; // for Arduino! Receiver on interrupt 0 => that is pin # 2
#endif

Do not forget to write back and tell me if it works.

Regards

mrbalintlorand, 2019-01-01:
Hi,

Thank you again for the help, unfortunately it still does not transmit for me with any of the tricks. :(

- I checked with the 8266 with swapping GDO0 and 2 as you suggested. I also made sure setGDO is not called (you're right, I saw that it would mess up GDO0 to be an input instead of an output as RCSwitch would set it...)
- I also went back again to try it with my ESP32. This is a WEMOS LOLIN32 with a built in oled, which uses GPIO4 and 5. So I connected the SPI wires to SPI2 with this added line in the code:
`ELECHOUSE_cc1101.setSpiPin(14, 12, 13, 15);`
I also figured out that this board also does not like GDO0 to be connected to GPIO0, as it never starts up that way. I tried connecting it to 2, 16, 25, 26, but no luck. :( Also of course confused on this board where to connect GDO2. As I googled around all these pins provide PWD.

Are we sure that the current latest master of the lib is the one others successfully set up with ESPs? I'll start diving in to git history, but tbh, I am a newbie to this programming area, so just slowly catching up.

Cheers,
Balint

mrbalintlorand, 2019-01-02:
I just ran in to this, and will try it out: Could it be that my problem is related to the ESP's Programmable PWM Frequency feature? And the default frequency might not be suitable?
Cheers,
Balint

LSatan, 2019-01-02:
So it works with esp.from me tested with esp32-wroom-32 chip and esp8266 wemos d1 mini. In another forum, I have received reports that it also works with nodemcu.

again to spi. pwm has nothing to do with the standard frequency. It is a communication bridge between the microcontroller and the cc1101, where data bits are exchanged to tell the cc1101 what to do. If you can receive with the esp then the spi communication works.

The driver makes that the gdo pins can now be used as with the simple rf modules. one to send one to receive.

It is strange that sending does not work for you.

Which cc1101 module do you use?
can you show me a picture or a link?

Regards

mrbalintlorand, 2019-01-02:
Hi,

This one:
https://www.aliexpress.com/item/CC1101-Wireless-Transceiver-Module-433MHz-2500-NRF-Distance-Transmission-Board-OOK-ASK-MSK-Modulation-Programable-Control/32855767133.html?tt=sns_none&aff_platform=default&cpt=1546433138567&sk=bCX9AWqc&aff_trace_key=73c7b0eb180649f3ac8743e2cbeca6b0-1546433138567-07531-bCX9AWqc&terminal_id=c8dcff6112754fbf831a1be889deab24

I seriously don't know what I am doing wrong. :(
I tried now with my other ESP8266 (also a NodeMCU), and receiving with the arduino, but no luck...
I unfortunately don't have more parts currently, but I might get some soon, and try additional setups...

Thanks for helping!
Cheers,
Balint

LSatan, 2019-01-03:
ok your module is a bit different. but should not be the problem.
After a few tests I had to find that the pins on esp 32 could not be changed.
I have updated the lib now. It is now possible to change the spi pins on esp32! I have seen in your oled version that the Spi pins are not available or used for the oled.

Try again with your esp32 with the following settings
ELECHOUSE_cc1101.setSpiPin(14, 12, 13, 15); .

SCK = 14, MISO = 12, MOSI = 13, CSN = 15.

I'm looking forward to your results.

Regards

mrbalintlorand, 2019-01-05:
Hi,

Thanks for the change! Now it works great with the ESP32, I could successfully receive and also send with it! :)
But unfortunately the ESP8266 is still not transmitting. :( I have the GDO0 and GDO2 swapped as you suggested in the comment above. I have another transmitter set up, those signals can be received, so receiving works, but transmitting does not.

Cheers,
Balint

LSatan, 2019-01-06:
that's great.
Then I'll order a nodemcu esp8266 to find the mistake. If I have new insights I will write it. But I'm not sure which version of nodemcu I will take.

In any case, thank you for your issue message. Only in this way can the lib be further improved.

regrads

LSatan, 2019-01-11:
Hello,
I ordered two nodemcu's.
the first arrived today and I tested it. It is a nodemvu v.0.9 with Esp 12 chip.
Everything is working fine! (with the specified pin settings from the esp8266.txt)

The second board is not there yet. It's a nodemcu v.3.2 with ESP 12-E chip. As soon as it has arrived I will test it and report the results here as well.

Regards

mrbalintlorand, 2019-01-11:
Hi,

Thanks for the followup. I hope we find my issue sooner or later. :)
Btw this is the 8266 board I was using:
https://web-engineering.info/sites/default/files/wot_tutorials/Mongoose-IoT-ESP8266/amica_nodemcu_v2.png

Cheers,
Balint

LSatan, 2019-01-12:
the second board arrived today. I tested it right away. It works fine too. I do not know what should be different on this board. Maybe it is broken !?
Which board from the boardloader do you choose for the upload?

regards

LSatan, 2019-01-16:
I have ordered the above cc1101 module. It is expected to arrive on 21.01 at my place.

balint-lorand, 2019-01-17:
Hi @LSatan 

I just tried the latest lib version, and now it works just fine transmitting on my ESP8266! :) (For the record, I kept transmitting on D2 to make sure that has PWM.)
Thank you very much for the continuous support and the fix!

Cheers,
Balint

mrbalintlorand, 2019-01-17:
Hi @LSatan

I just tried the latest lib version, and now it works just fine transmitting on my ESP8266! :) (For the record, I kept transmitting on D2 to make sure that has PWM.)
Thank you very much for the continuous support and the fix!

Cheers,
Balint

closed 2019-01-17 by mrbalintlorand