LSatan/SmartRC-CC1101-Driver-Lib #11: Strange behaviour ESP12e

sanschaussure, 2019-01-11:
Hello,

I set up this cc1101 module 

https://www.wantitall.co.za/electronics/solu-cc1101-wireless-transceiver-module-with-spring-antenna-for-arduino-wireless-rf-transceiver-315-__b00xdl9e64

And found two curious things:

**First:** 
My setup with an Arduino UNO (SPI level 5V, module survives) with the simple receive example worked out of the box. I received a lot of signals from my different transmitters (24bit, 32bit).

I further on did the same with an Wemos D1 mini (clone, SPI level 3.3V). Setup was like in the example image. The module only received a few 32bit signals. Also the decoded numbers didn't belong to one of my transmitters and I havn't seen them in the first Arduino test (so no signals from the neighbors).

To test the program I replaced the CC1101 on the Wemos by a standard cheap RF433 receiver. It worked. So the Wemos is capable to receive signals. Then I added this line to the setup of the program: 

`SPI.setHwCs(true);`

Now I received 24bit and 32bit messages. But not as much messages as with the Arduino system. 

Then I mixed up the systems. Connecting the SPI of the CC1101 module to the Arduino and the signal out to the Wemos. Now the Wemos receives all messages and the same amount of them - like the pure Arduino system.

_Now my question? Is there a difference in the initialization of the module between ESP and Arduino boards. I personally haven't found differences in the lib code._

**Second**
My seconds observation is curious, too. In my setup I have two types of transmitter modules. Both are working with the standard receivers.

The setup is 10 of the more expensive type:

https://es.aliexpress.com/item/Wholesale-433-Mhz-Superheterodyne-RF-Receiver-and-Transmitter-Module-ASK-kits-with-antenna-For-Arduino-Diy/32799575156.html

and 3 of the cheap ones:

https://www.electrodragon.com/product/433m-rf-wireless-transmitter-module/

In the working pure Wemos setup the signals of the cheap modules were not received. In the Arduino setup signals from every module were received.

_Question: Are there differences in the signal recognition depending on the used MCU?_

LSatan, 2019-01-12:
Hello,
I do not understand the problem completely. my English is not the best. First to your first question:
The Spi setup of the Elechhouse cc1101 lib has been adapted to work with arduino and esp.
the only problem I know, which will be fixed in future updates, is that the esp module does not receive any messages from the serial monitor and also not from the rx pin of the esp's when spi was initialized.

They said they used an arduino to address the cc1101 via spi. and the gdo pins to receive / send to have an esp connected and that it worked better than if you control the cc1101 module on esp with spi. it may be possible that it works better. I dont know.

I use esp wemos d1 mini for rc-switch and have no problem. So I receive the same as when I use the cc1101 module on arduino.
so it would be good to know which sketch you use to receive with the esp !?

for your second question:
I personally have found that the pins of the esp's are more sensitive than those of the arduino's.

in general, it is a lib that is in the process of development and is constantly being improved. I also use it for raw data that you can receive via RF (but with an arduino).

So to understand the problem! Which sketch do you use?

I will test it then!

Regards

sanschaussure, 2019-01-13:
I will explain the problem above in German. Maybe it's the easier to solve.

Ich habe das in den Bildern gezeigte Arduino und Wemos-Setup aufgebaut. Exakt wie im Bild. Immer den gleichen CC1101. Als Sketch "ReceiveDemo_Simple_cc1101" auf beide Systeme geflasht. 

Der Arduino-Aufbau hat ohne Probleme funktioniert. Der Wemos-Aufbau erst, als ich die Code-Zeile

`SPI.setHwCs(true);`

und den Flash-Mode auf DIO gestellt habe. Dazu in der IDE auf "Generic ESP8266 Module" gestellt. Problem des Wemos-Aufbaus ist, dass er viel weniger Signale empfängt (z.B. von 30 Signalwiederholungen nur 10) und bei Signalen mit geringerer Wiederholung auch mal gar keines. Zudem "sieht" der Wemos-Aufbau keine Signale der ganz billigen Sender - was der Arduino-Aufbau tut.

Um einen Fehler im Wemos auszuschließen, habe ich ihn mit einem normalen Superheterodyn-Empfänger betrieben. Hat ohne Probleme mit dem Sketch funktioniert. 

Danach habe ich den CC1101 mit dem Arduino betrieben, den Signalausgang aber auf den Wemos geleitet. Auch damit hat der Wemos sehr gut funktioniert. Ein Bild dazu ist hier, links der Arduino mit CC1101 uns nach rechts Signal-Out und GND zum Wemos:

[image]

Meinem Eindruck nach gibt es irgendwelche Unterschiede in der Initialisierung des CC1101 zwischen den Systemen. Grundsätzlich funktioniert der CC1101 auch am Wemos, aber nur sehr selektiv. Ich habe das auch mit einem normalen ESP12e nachgestellt. Das ergab das gleiche Bild.

Ich habe gestern auch noch festgestellt, dass der CC1101 wesentlich weniger sensitiv ist als ein einfacher Superheterody-Empfänger. Auch am Arduino.

LSatan, 2019-01-13:
Hi

Ich habe ein bisschen verglichen. Hier die ergebnisse:

Test 1
Esp + CC1101 VS. Nano + CC1101

[image]

[image]

Der esp reagiert teilweise schneller.
Ich tendiere aber mehr zu unendschieden.

Test 2
Esp + CC1101 VS. Nano + Superheterodyne Receiver

[image]

[image]

Auch kein großer unterschied.

Rein technisch haben die Superheterodyne Receiver eine empfangs bandbreite von ca 2000khz. Der cc1101 hat mximal 812khz.
Also nehmen wir mal wir haben eine frequenz von 433,92 mhz ausgewählt. Dann liegt der empfangsbereich zwischen 433,514mhz und 434,326 mhz.
Sollte der Sender nicht innerhalb dieser frequenzen senden dann wird es vom cc1101 nur schlecht erfasst.
Hier kann man nur die frequenz entsprechend des Senders anpassen.

Ich denke aber das wir so nicht weiter kommen um das problem zu lösen.
Ich kann die sketche so verwenden wie sie sind und alles läuft gut.

Warum nutzt du zum upload Generic Esp8266 module? Du solltest Wemos D1 R2 & mini auswählen.
Die einstellungen müssen dort nicht verändert werden und der upload ist wesentlich schneller!

Vieleicht sollten wir nochmal die software abgleichen!
Ich nutze Arduino IDE 1.8.5
und die Boardloaderversion: Esp8266 Boards (2.5.0-beta2).

Übrigens sieht deine antenne stark nach 866MHZ aus!?

Wenn es nicht an der software liegt dann vieleicht an deiner cc1101 version.

Ich hoffe wir finden eine lösung für das problem!

Grüße

LSatan, 2019-01-14:
I have updated to ide 1.8.8. no difference. everything works. So maybe the cc1101 module. I will order this version for tests.

Regards

LSatan, 2019-01-15:
OK. Too bad that it does not work right for you. The CC1101 module has a lot of potential. Thanks for the issues message. I'll take a look where the problem is.

regards.

sanschaussure, 2019-01-21:
Thank you for the fix. Now both receivers work similar. 

Except I recognize that the range of my cheap transmitters is bigger compared to the more expensive ones. But on the other hand for the cheap ones I have to choose a higher filter bandwidth due to less accurate transmitting frequencies.

closed 2026-04-26 by github-actions[bot]