LSatan/SmartRC-CC1101-Driver-Lib #12: mySwitch.available() is never true but I receive something

garytube, 2019-01-13:
My Setup is a nodemcu ESP32 Devkit v1 and a CC1101by D-SUN

I followed the wiring provided in the repo (thx!) and when ever i press a button on my remote the built-in LED flashes. So apparently I receive something but the function mySwitch.available() never evals to true.

What am I'm missing?

I added a simple debug statement which fires everytime I press a button on my remote.
`
void loop() {
  if(digitalRead(pin) == 1){
    Serial.println("got something");
  }
...
`

LSatan, 2019-01-14:
Hi
If the LED is flashing, the CC1101 module will work fine. The error indicates that rc-switch does not know the protocol of your remote control. I will try to help you. Which remote do you use?

Regards

LSatan, 2019-01-14:
Does your cc1101 module have 8 pins or 10 pins?

garytube, 2019-01-14:
My Module has 8 Pins.
Will try to figure out what model my remote is. I believe I need to crack one open :)

LSatan, 2019-01-14:
do not do that! First try my tip:
delete rc-switch and replace it with this version:

https://github.com/Martin-Laclaustra/rc-switch/tree/protocollessreceiver

and for cc1101 use this version of ProtocolAnalyzeDemo:

[ProtocolAnalyzeDemo_cc1101.zip](https://github.com/LSatan/RCSwitch-CC1101-Driver-Lib/files/2755110/ProtocolAnalyzeDemo_cc1101.zip)

If it works, please give feedback.

regards