LSatan/SmartRC-CC1101-Driver-Lib #14: Will you port this library for Raspberry or is there any simple way to make it work?

YindSoft, 2019-03-26:
This is exactly what I need, but I have to use a Raspberry.

There is a library of CC1101 for Raspberry, but I don't get how to make it work this way to receive the ids of the rf433 devices.

Any idea how?

LSatan, 2019-03-27:
Hello, A port would be possible. but currently have little time. Alternatively, you could connect an arduino nano to the cc1101 and then let the nano talk to the pi. Can you post a link of the librarie that works with the pi? I would then take care of the rest. If it is possible. earliest next week. Regards

YindSoft, 2019-03-27:
This is the library: https://github.com/SpaceTeddy/CC1101

Thank you!

TMInnovations, 2019-04-28:
Firstly, thank you for the great work on the Driver library for Arduino & Co.!

I'm looking for the exactly same solution as @YindSoft is. With it the CC1101 will get the best piece of Transceiver by far for switching plugs & Co. on the RPi!
Hope it's possible for you to get it working! ;-)

Thank you in advance!

normen, 2020-04-18:
This library is basically made for bit banging switch codes and that doesn't work very well on RasPi, I'd go with the suggestion and use a separate microcontroller and just connect it via USB to the Pi. Heres a project of mine that does exactly that to control cheap 433MHz switches: https://github.com/normen/arduino-433

Using the CC1101 for communicating on RasPi works, but only if you let the CC1101 use it's internal protocol to send and receive data. Then it takes the burden of decoding off the main computer and just delivers the data. But then you can't control 433 switches. Thats why basically all home control systems went the external microcontroller / USB way.

fphammerle, 2020-08-23:
potential alternative for raspberry pi: https://github.com/nahueldsanchez/pycc1101 (python)

fightforlife, 2020-10-23:
Is there any news on the PI port?
I am currently using a ESP32 with this library. But the ESP32 has not enough oompf to handle Wifi, BT, MQTT, CC1101, ... at the same time. I really would like to switch to a Pi4.
I will look into using a Arduino Nano as bridge in the mean time.
But a port of the library would be perfect since it is using the one of the best transmitters with the best 433mhy library rcswitch.

Thanks in advance!

LSatan, 2020-10-24:
Hi, i agree that porting to rpi is an interesting thing. Unfortunately, I haven't dealt with the raspberry enough until now. because I don't have the time to do a port. I assume that a control via the thermal program meets the requirements. I found the following interesting approach:
https://www.deviceplus.com/raspberry-pi/how-to-run-arduino-sketches-on-raspberry-pi/

 it may be necessary to make adjustments to the bibiliotehk with regard to the gpio pins. share your opinion with me.
Regards

closed 2026-04-26 by github-actions[bot]