LSatan/SmartRC-CC1101-Driver-Lib #15: Short transmission distance for the frequency of 915 MHz.

RomanPetrenko, 2019-04-21:
Hey.
I bought this module here. 
https://uamper.com/%D0%9C%D0%BE%D0%B4%D1%83%D0%BB%D0%B8/%D0%91%D0%B5%D0%B7%D0%B4%D1%80%D0%BE%D1%82%D0%BE%D0%B2%D1%96/433-%D0%9C%D0%93%D1%86/CC1101-Wireless-Transceiver-Module-with-SMA-%D0%BF%D1%80%D0%B8%D0%B9%D0%BC%D0%BE%D0%B4%D0%B0%D0%B2%D0%B0%D1%87

I programmed it for the frequency 433.92 MHz - everything works fine. I changed it to 915 MHz - it works at a distance of up to 1 meter. Then the signal is interrupted, what can it be?

LSatan, 2019-04-24:
Hello firstly the antenna is designed for 433 MHz. Secondly. your module too. There is an extra 868 MHz module with which you can also send in the 915 MHz range. There, the capacitors and the resistors are slightly different antennas. Basically, the 433mhz module can send in this area but the result you know. whether it is better with a 868mhz antenna I do not know.

regards

b-ganesh-rao, 2020-01-16:
hi , 
I have CC1101 868Mhz module with 8 pins on one side and 3 pins (including antenna) on the other side .  Doesn't respond  for identifying protocol  using example code  ProtocolAnalyzeDemo from  rc-switch-protocollessreceiver library.
please help me to debug the issue . All the wirings to nodemcu are mapped as per the diagram  in the code.

regards
B.Ganesh

LSatan, 2020-01-16:
Hi Which handheld transmitter do you use?

LSatan, 2020-01-16:
most devices that transmit on this frequency use a different modulation. it is very rare that ask is used at these frequencies. The library will soon receive an update. other modulations are then also supported there. it will not run with rc switch. a new library must be written.

b-ganesh-rao, 2020-01-17:
Hello , 
Its the remote for my curtain motor .(chinese make). N18 curtain motor . The remote supplied along with motor ,I checked with RTL SDR for the protocol and the frequency . Its 868 MHz FSK modulation. whats the best approach to  record and simulate this remote 

Thanks and Regards
B.Ganesh

LSatan, 2020-01-19:
Hi there Sorry for the late reply. I'm currently looking for a way to get raw data from fsk. in general it should be possible. Some tests have shown that sending is similar to ask with digitalwrite. only that both pins gdo0 and gdo2 must be used here. it is similar when receiving. if there is something new I will let you know. regards

b-ganesh-rao, 2020-01-20:
Thanks LSatan for the reply , I have tried the other day alternative
solution for the issue ( instead of mimic the remote ) , wanted to mimic
the switch using fauxmosesp library , that has issues with compatibility
with echodot versions.
Anyways, I like to take your services not for free but for some fee ,
please let me know your fee for the workable solution.

Thanks
B.Ganesh

closed 2026-04-26 by github-actions[bot]