LSatan/SmartRC-CC1101-Driver-Lib #16: ShowRecievedCode works but not RecieveDemo

AlexanderKlasson, 2019-10-25:
Hi,
To begin with, I am a beginner at this. 

I am trying to use an D-SUN CC1101 together with a Wemos D1 mini to control my NEXA power outlets (433.92MHz). When I run the script ShowRecievedCode I get the following:

Code: 1785846 Period: 257
 unit: 0
 groupBit: 0
 switchType: 1
Code: 1785846 Period: 256
 unit: 0
 groupBit: 0
 switchType: 0
Code: 1785846 Period: 256
 unit: 1
 groupBit: 0
 switchType: 1
Code: 1785846 Period: 257
 unit: 1
 groupBit: 0
 switchType: 0

But when I run the script RecieveDemo (both simple and advance) I recieve nothing. 

Do you know what could be the reason behind this? My friend is trying the same thing with the same setup and has the same result. 

Thank you for your contribution on this!

LSatan, 2019-10-25:
Hi,

receive demo simple / advanced (rc-switch) is for the simple radio protocols and will not work with your remote control.
that's why there is new remoteswitch. which then works with your remote control. to send then you use light show of new remote switch. with remotes fusion you can use rc switch and new remoteswitch at the same time. Android phone is required. it will also be an update again what a bug with the connection to the router fixes. Currently I am working on a library that will support many remote controls. release may take some time. For further questions, please do not hesitate to contact me.
Thank you for your interest.

regards

AlexanderKlasson, 2019-10-25:
Ahh I see!

It seems to be working for me, thank you!

closed 2026-04-26 by github-actions[bot]