LSatan/SmartRC-CC1101-Driver-Lib #21: Two receivers with two CC1101 at the same time

joelsernamoreno, 2020-01-06:
Hello!

Is it possible to use 2 CC1101 as 2 receivers working at the same time? Example: 1 CC1101 at 433mhz/1 CC1101 at 315mhz

I've seen that you can change modules with ELECHOUSE_cc1101.setSpiPin, but I don't know if when I change to the second module I'll lose the reception of the first module

Example:

ELECHOUSE_cc1101.setSpiPin(10, 11, 12, 13);
ELECHOUSE_cc1101.setMHZ(433.92);
ELECHOUSE_cc1101.Init(PA10);
ELECHOUSE_cc1101.SetRx();
mySwitch.enableReceive(0);
ELECHOUSE_cc1101.setSpiPin(10, 11, 12, 13);
ELECHOUSE_cc1101.setMHZ(315.00);
ELECHOUSE_cc1101.Init(PA10);
ELECHOUSE_cc1101.SetRx();
mySwitch.enableReceive(1)

Thanks!

LSatan, 2020-01-06:
Hi there,

You may lose the connection to the first one (due to the RC switch).
The settings in the CC1001 modules remain.
I would suggest testing the following:
https://www.taillieu.info/index.php/software/arduino-sketches/287-rf-sniffer-rx433-tx433

Not tested itself. I would appreciate feedback.

Regards

joelsernamoreno, 2020-01-06:
I tried this last night, but it didn't work:

https://z4ziggy.wordpress.com/2014/06/27/rf-sniffer-open-gates-cars-and-rf-remote-controlled-devices-with-ease/

I'll keep running tests
Thanks!

LSatan, 2020-01-06:
at the end try to initialize rc-switch pins.

ELECHOUSE_cc1101.setSpiPin(10, 11, 12, 13);
ELECHOUSE_cc1101.setMHZ(433.92);
ELECHOUSE_cc1101.Init(PA10);
ELECHOUSE_cc1101.SetRx();
ELECHOUSE_cc1101.setSpiPin(10, 11, 12, 9);
ELECHOUSE_cc1101.setMHZ(315.00);
ELECHOUSE_cc1101.Init(PA10);
ELECHOUSE_cc1101.SetRx();
mySwitch.enableReceive(0); //Pin D2
mySwitch.enableReceive(1) //Pin D3

LSatan, 2020-01-06:
assign unique names for the modules:

RCSwitch rf315Switch = RCSwitch();
RCSwitch rf434Switch = RCSwitch();
and:
 rf315Switch.enableReceive(0); 
 rf434Switch.enableReceive(1); 

I wish you success

joelsernamoreno, 2020-01-06:
I'm using an ESP32 module

My pinout is as follows

GPIO ESP32 / 1 CC1101
------------------------------------
3V3 (3.3V) / VCC
G (GND) / GND
18 (SCK) / SCK
19 (MISO) / MISO
23 (MOSI) / MOSI
5 (SS) / CSN
2 (PWM) / GDO0 (TX)
4 (PWM) / GDO2 (RX)

GPIO ESP32 / 1 CC1101
------------------------------------
3V3 (3.3V) / VCC
G (GND) / GND
18 (SCK) / SCK
19 (MISO) / MISO
23 (MOSI) / MOSI
15 (SS) / CSN
34 / GDO0 (TX)
35 / GDO2 (RX)

I think the pinout of the second module is wrong, in theory all digital pins of ESP32 allow interruptions, but this doesn't work for me, only the first module works.

This is the code:

 ELECHOUSE_cc1101.setSpiPin(18, 19, 23, 5);
  ELECHOUSE_cc1101.setMHZ(433.92);
  ELECHOUSE_cc1101.Init(PA10);
  mySwitch.enableReceive(4);
  ELECHOUSE_cc1101.SetRx();

  ELECHOUSE_cc1101.setSpiPin(18, 19, 23, 15);
  ELECHOUSE_cc1101.setMHZ(315.92); 
  ELECHOUSE_cc1101.Init(PA10);
  mySwitch.enableReceive(35);
  ELECHOUSE_cc1101.SetRx();

This is a simple code to change the module (first I am testing that the module change works correctly), but I think that the digital pins 34 and 35 are not the correct ones. Can you indicate which are the correct pins for the second module CC1101 (TX and RX)?

Thank you very much :D

LSatan, 2020-01-07:
it is a bit difficult with the esp32.
the pins for sending are not necessarily the problem.
with the exception of gpio 34.35 as far as I know they are only entrances.
You can at most test the pins to receive them.
GPIO 3 definitely functions as a pin for receiving. I tested it myself. that's the rx pin.
works even if a second cc1101 is on GPIO 4. The only problem with the rx pin is that you have to disconnect the cable from the cc1101 to the rx pin before flashing.
I think there are other pins. for me, however, no more keep since I still have an SD card reader and a display on it.

joelsernamoreno, 2020-01-07:
I just disconnected VCC and GND from the second CC1101 module and it works again with the first module 

I think connecting the two modules with the pinout I told you about breaks everything

joelsernamoreno, 2020-01-07:
Thank you

I'll also be testing with Arduino Nano

I'll write to you if I can make it work

Thank you very much!

LSatan, 2020-01-07:
dies sind meine pins:
sck = 14;
miso = 12;
mosi = 13;

cc1101 / 1
gdo2 GPIO4
gdo0 GPIO2
csn = 15

cc1101 / 2
gdo2 GPIO 3 (RX)
gdo0 GPIO 16
csn = 27

joelsernamoreno, 2020-01-07:
Thank you!

Tomorrow I'll try this

I'm building a PCB with the electronics for a complete device.

When I have this, you will have a free device

joelsernamoreno, 2020-01-12:
When I connect the second module, everything locks. If I disconnect the second module, it works again

Pinout:

Arduino Nano / CC1101 (1)
3.3     ->     VCC
GND  ->    GND
D13   ->    SCK
D12   ->    MISO
D11   ->    MOSI
D10   ->    CSN
D6     ->    GDO0 (TX)
D2     ->    GDO2 (RX) Interrupt 0

Arduino Nano / CC1101 (2)
3.3     ->     VCC
GND  ->    GND
D13   ->    SCK
D12   ->    MISO
D11   ->    MOSI
D9   ->    CSN
D5     ->    GDO0 (TX)
D3     ->    GDO2 (RX) Interrupt 1

joelsernamoreno, 2020-01-12:
Advance:

I've added three 10k resistors in miso, mosi and sck (pull up), this works. When I connect the second module, the first module works.

Tomorrow I will try to change module and the tests we talked about a few days ago 

Thank you!

joelsernamoreno, 2020-01-12:
I can switch modules, but I can't enable two receivers at the same time

The problem is the setRX function, that function enables the receiver in the last module you have configured

If I separate this into 433Swith and 315Swith (the link you sent me), this doesn't work either because of the setRX function

LSatan, 2020-01-12:
Hi there I will test it myself the days. I have enough modules. which cc1101 modules do you use? the dsun 8pin or the other 10pin?

joelsernamoreno, 2020-01-12:
10pin:

https://www.amazon.es/neuftech-CC1101-Wireless-Module-Transceptor-464-MHz/dp/B01CI01F94/

Thank you!

I can change module and enable a receiver, but the setRX function chooses the last module configured, the other module disables it

Thank you very much!

LSatan, 2020-01-13:
I have tested it.
only received with 2 modules. it works well.
2 cc1101 on the esp32 and 1 cc1101 module on the arduino nano as transmitter. the transmitter always transmits alternately between 315 and 433 MHz. everything is received.
send on 315 mhz
Received 3152 / 24bit Protocol: 1
Received 3153 / 24bit Protocol: 1
Received 3153 / 24bit Protocol: 1
Received 3153 / 24bit Protocol: 1
Received 3153 / 24bit Protocol: 1
Received 3154 / 24bit Protocol: 1
Received 3154 / 24bit Protocol: 1
Received 3154 / 24bit Protocol: 1
Received 3154 / 24bit Protocol: 1

send on 433mhz
Received 4331 / 24bit Protocol: 1
Received 4331 / 24bit Protocol: 1
Received 4331 / 24bit Protocol: 1
Received 4332 / 24bit Protocol: 1
Received 4332 / 24bit Protocol: 1

when receiving at 315 mhz the onboard led does not flash. at 433 mhz already
here is my sketch:

`

#include <RCSwitch.h>
#include <ELECHOUSE_CC1101_RCS_DRV.h>

//CC1101 Pins
int sck = 14;
int miso = 12;
int mosi = 13;
int csn1 = 15;
int csn2 = 27;
int gdo0_1 = 2;
int gdo2_1 = 4; 
int gdo0_2 = 16;
int gdo2_2 = 3;

float CC1101_1_Freq = 433.92;
float CC1101_2_Freq = 315.00;

RCSwitch mySwitch = RCSwitch();
RCSwitch CC1101_1 = RCSwitch();
RCSwitch CC1101_2 = RCSwitch();
 
void setup()
{

  Serial.begin(115200);
  //setup cc1101
  ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn1);
  ELECHOUSE_cc1101.setMHZ(CC1101_1_Freq);
  ELECHOUSE_cc1101.Init(PA10);
  CC1101_1.enableReceive(gdo2_1);
  ELECHOUSE_cc1101.SetRx();
  
  ELECHOUSE_cc1101.setSpiPin(sck, miso, mosi, csn2);
  ELECHOUSE_cc1101.setMHZ(CC1101_2_Freq);
  ELECHOUSE_cc1101.Init(PA10);
  CC1101_2.enableReceive(gdo2_2);
  ELECHOUSE_cc1101.SetRx();
  
}
 
 
void loop()
{
 
        if (mySwitch.available()) {
        
        Serial.print("Received ");
        Serial.print( mySwitch.getReceivedValue() );
        Serial.print(" / ");
        Serial.print( mySwitch.getReceivedBitlength() );
        Serial.print("bit ");
        Serial.print("Protocol: ");
        Serial.println( mySwitch.getReceivedProtocol() );

    mySwitch.resetAvailable();

}
}
`

joelsernamoreno, 2020-01-13:
Oh, you have three RCSwitches; I had only created two of them

In the loop, I have the following:

If (rf433.available()) {
Command...
}

else if(rf315.available()) {
Command...
}

I'll try your code tomorrow, thanks!

joelsernamoreno, 2020-01-13:
Hello...

I tried your code on arduino Nano (changing pins) and ESP32, but it doesn't work for me. It only works with 433mhz. It's not an error of my 315mhz transmitter, I've made tests and I can receive it if I only use a RX.

What additional hardware do you use? Right now I have three 10k resistors (pull up), if I don't have these resistors, the modules don't have enough power

I'll keep testing in the meantime, thank you!

LSatan, 2020-01-13:
I do not use additional resistors or the like. after flashing on the esp32, i reset it or briefly take the current away. then it works. how far are you from the esp32? it is so that if both cc1101 receive at the same time there is no output. Your signal may break through if you hold the transmitter too close.

joelsernamoreno, 2020-01-13:
Hello

I've got it working on Arduino Nano
I will continue with ESP32 :D

Thank you very much

LSatan, 2020-01-16:
Hi there, have you got any further? I have found that something is received at both times only at one. a change in the cc register helps and it always works. it will be included in the next update.

joelsernamoreno, 2020-01-16:
Hello!

With Arduino Nano I have no problem

I'll make this change in the registers and try ESP32

Thank you!

joelsernamoreno, 2020-01-30:
Well, I've changed this. So far, everything's working fine.

I'll run tests for another week, if everything works out, we'll do the PCB design

orrpan, 2020-05-01:
Could you have four?
315
433.92
868
915

LSatan, 2020-05-01:
yes this is theoretically possible. also depends on how many pins the microcontroller used has. Another question is whether rcswitch supports this. but with the lib you can switch between several cc1101s. it is up to you whether the whole thing makes sense. regards

joelsernamoreno, 2020-12-24:
Hello,

I have updated to your latest version of the SmartRC library

Well... I am using an ESP32-PICO and two CC1101 modules

If I set the CPU frequency of the ESP32-PICO to 40 MHz (40 Mhz XTAL) both CC1101 modules work at the same time, but if I set the CPU frequency to 80 MHz only 1 CC1101 module works

To set up a WiFi access point, ESP32 needs 80 Mhz of CPU frequency

Well, I have changed SpiWriteReg (CC1101_FSCTRL1, 0x06); to
SpiWriteReg (CC1101_AGCCTRL2, 0x07); but now no module works

By the way, I already have prototypes of Evil Crow RF, when this works properly I will send you one:

[image]

Thank you!

joelsernamoreno, 2020-12-26:
Hello!

I downloaded version 2.3.0 of your library, I changed SpiWriteReg(CC1101_FSCTRL1, 0x08) to SpiWriteReg(CC1101_FSCTRL1, 0x09) and both CC1101 modules work

I tried to do this in the last version and it doesn't work. What is the SpiWriteReg equivalent to what I told you in the last version? Thank you!

LSatan, 2020-12-26:
hi, FSCTRL1 is responsible for the following. for the calculation one should look into the data sheet:

The desired IF frequency to employ in RX. Subtracted from FS base frequency in RX and controls the digital complex mixer in the demodulator. (See datasheet for equation). The default value gives an IF frequency of 381kHz, assuming a 26.0 MHz crystal.

joelsernamoreno, 2020-12-27:
Thank you! For now I will use the old version of your library 

I have given you access to the Evil Crow RF repository in case you want to add new features or use the code for yourself

This will be public when Evil Crow RF is on sale, I promise it will be very cheap :)

LSatan, 2020-12-27:
thanks for the invitation, is there a specific reason for using 2.3? The version still has quirks in the frequency calculation. Which major construction sites are still open? You can best reach me directly via my email. littlesatan.666.ls@gmail.com
Regards

gc9n, 2023-11-05:
> SpiWriteReg (CC1101_AGCCTRL2, 0x07);

Hi there  , 
i just swap this 

float CC1101_1_Freq = 315.00;
float CC1101_2_Freq =433.92 ;

and i still get the 433 in the #1 cc1101  , how is that possible its mind blowing