LSatan/SmartRC-CC1101-Driver-Lib #23: Separating other code and apks

normen, 2020-02-12:
When including this library in PlatformIO via github a lot of unneeded stuff gets downloaded to the library folder. Theres apks with Android apps (?) and all other kinds of stuff in this repo that doesn't really belong to the library.

I wondered if it was possible to move these things to a different repository so that this repo can be used solely for the library and its documentation.

Again, this is your project and you can of course do what you want with it but for the sake of simplicity and to avoid confusion if this library ever gets listed in the arduino library manager I suggest keeping the repo clear of external stuff.

Cheers,
Normen

LSatan, 2020-02-19:
Hello Normen,

Thank you for your letter.
Sorry for the late reply, I only saw your message today.
I am always open to praise, criticism and suggestions for improvement.
Unfortunately, you can't always please everyone.

When I bought the cc1101 module some time ago,
I did not assume that it would not work with the RC switch.
So I got to work to make it work for me.
After a short time it was ready to receive and send.
Now I wanted to share my results with everyone, show something with this
module is possible, especially that it just works.

Over time, a few things have been added, improvements, etc.
Some people use things there that others may not need.
Usually it shouldn't be rc-switch_cc1101_driver_lib anymore,
since it should work in theory with all libraries that use a "cheap" ASK module.

Let's take Issues # 16 as an example.
Here someone tried to receive and send something with Rc-switch which is not compatible with rc-switch.
So I referred him to new remote switch and it worked for him.

Regardless of your letter, I have been thinking about restructuring the library for quite a while
or to rename it as it is no longer up to date and in the next update too
there are a few things that are similar to the settings of a smart rf studio.
I don't know enough about github to use all functions to their full extent.
Because everything just takes too much time.
I will continue to think a little bit and I also like to work on expanding the library.

excuse my bad English
Kind regards LSatan

normen, 2020-04-19:
I see the library has been cleaned up, very nice! 👍

closed 2020-04-19 by normen