LSatan/SmartRC-CC1101-Driver-Lib #24: GDO2 incorrectly set to INPUT

jeffkowalski, 2020-03-23:
In the recent updates, the GDO_Set function was altered to set pinMode(GDO2, INPUT); Since it's a signaling pin from the driver to the CC1101, it should be set to pinMode(GDO2, OUTPUT); as it was before the updates.

the last working version was https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/a340750b595135babd49fcc701f7ebceca1c4e2f/ELECHOUSE_CC1101_RCS_DRV.cpp

LSatan, 2020-03-24:
Hi there, 
the original elechouse lib uses both gdo pins as input.
 I set it to output at some point. Why does the devil know. 
Theoretically, the pins do not have to be defined at all. Then you would automatically be input. 
I could put the pin back to output. it is not in conflict.
Do you have an application that no longer works due to the change? Or was it just a statement that the pin was changed? if the first one is correct I will change it as soon as possible.
 regards

jeffkowalski, 2020-03-24:
@LSatan, https://github.com/EinfachArne/Somfy_Remote will not work unless
GDO2 is an output pin.  It pulses that pin directly
https://github.com/EinfachArne/Somfy_Remote/blob/master/src/Somfy_Remote.cpp#L149

LSatan, 2020-03-24:
ok i remember. I will change it and test it again. update is still coming today. European Time.
Thanks for your message.
regards

LSatan, 2020-03-25:
in the lib of einfach arne would still have init (pa10); on init (); be changed. since the command is only there for the initial one. the best thing is to simply write that it will be changed.
regards

jeffkowalski, 2020-03-25:
Yes, thank you for that hint.  I had already changed to ```init();``` in my
local copy.  I will file an issue with the author @EinfachArne.