LSatan/SmartRC-CC1101-Driver-Lib #25: worse RX range than in old library (ELECHOUSE_CC1101_RCS_DRV)

fightforlife, 2020-03-29:
Hey there,

I was using your old library for quite some time. (ELECHOUSE_CC1101_RCS_DRV)
I love it and it works perfectly with RCSwitch on an ESP32.

In my sketch I replaced the old library with the new ELECHOUSE_CC1101_SRC_DRV.

Since then I have much worse RX signal strength. The CC1101 is only receiving the correct signals when I hold the remote directly in front of it.
With the old library I could see the commands of my neighbours remote.

These are the things I changed. I can reproduce it easily by changin these lines.

old sketch:
```
#include <ELECHOUSE_CC1101_RCS_DRV.h>
...
void setup(void) {
...
  ELECHOUSE_cc1101.Init(12);
...
}
```

new sketch:
```
#include <ELECHOUSE_CC1101_SRC_DRV.h>
...
void setup(void) {
...
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.setPA(12);
...
}
```

LSatan, 2020-03-29:
Hello,
Thank you for your message.

If you

ELECHOUSE_cc1101.Init ();
ELECHOUSE_cc1101.setPA (10);

puts,
is it working properly again?

regards

LSatan, 2020-03-29:
oh these are your changes and it is going badly. sorry i misinterpreted that. I am currently creating a debugging tool and will review it as soon as possible. thanks again. regards

LSatan, 2020-04-01:
problem is solved with v2.3.4.

regards

fightforlife, 2020-04-02:
That was really quick! Thank you, receiving is much better now!

closed 2022-06-13 by fightforlife