LSatan/SmartRC-CC1101-Driver-Lib #26: Wrong Image Pinout for ESP32

minkione, 2020-04-17:
According to your image the pinout is not matching the usual layout of an ESP32-wroom
https://circuits4you.com/wp-content/uploads/2018/12/ESP32-Pinout.jpg
VS
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/img/Esp32_CC1101.png

LSatan, 2020-04-17:
Hello,
the representation is a NodeMcu-32S with wroom esp32 chip. 
https://cdn.instructables.com/FOL/YWLI/JEOILQ5U/FOLYWLIJEOILQ5U.LARGE.jpg?auto=webp&frame=1&width=1024&fit=bounds

I chose this model for the illustration because the lettering is from below and is therefore difficult to see. The pictures are only examples and one should refer to the labeling between the cc1101 and the controller. Finally, you can also connect an esp8266 nodemcu with the displayed pin assignment instead of an esp8266 wmos d1 mini.
Thank you for the message, I will add the controllers shown to the pictures to avoid confusion.
regards

minkione, 2020-04-17:
Ok then make sense. Is another board.
Since there was not any specific name of the model I inferred was still the WROOM one.
So in case to avoid confusion to users, I preferred pinging you. :)

closed 2020-04-17 by minkione

minkione, 2020-04-17:
Just had a thought...
I just noticed these new examples https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master/examples/CC1101%20default%20examples

That means that now the library do support sending/receiving data with different modulations (FSK, etc) besides the usual ASK?

LSatan, 2020-04-17:
yes you can set the other modulations. when you send from module to module.
 With rc-switch it makes no sense, of course, since lib only supports ask modulation.

LSatan, 2020-04-17:
of course it also depends on what you intend to do with it.

minkione, 2020-04-17:
Cool.  Thanks for the confirmation.
Another thing... Mega2560's pins run at 5V.
CC1101 instead logic runs at 3V3.

I guess a level shifter is needed between them.

LSatan, 2020-04-17:
i also use the cc1101 module with an arduino nano. which also has 5v logic. the modules have been in use without a logic converter since 2018. Everything still works without anything breaking. But if you want to be on the safe side, you can hang a logic level converter in between. or corresponding voltage dividers over resistors. but as I said, nothing has ever been damaged.

minkione, 2020-04-17:
Yeah me too, never had issues.
Though I though would be wise to advise users anyway. Like mentioning it somewhere. ;)

LSatan, 2020-04-17:
yes of course that's true