LSatan/SmartRC-CC1101-Driver-Lib #27: Is the GDO-2 really needed to be wired?

minkione, 2020-04-18:
Hi,
I couldn't find any example where the GDO-2 is really needed.
Is it really needed to be wired?

LSatan, 2020-04-18:
Hello again.
Gdo-2 is used in the examples for reception (except for the internal reception method). Yes, you can also use gdo0 for both. just like it was in the first versions of lib (before esp8266 support). However, since the esp did not cope with the fact that one suddenly turns the receive pin into a send pin gdo2 was added. and all was fine.

LSatan, 2020-04-18:
i think it is also easier for the end user to program if there is one pin to send and one to receive.

minkione, 2020-04-18:
Thanks for the quick reply. 
Perfectly agree on making easier end-user life. ;]

closed 2020-04-18 by minkione

LSatan, 2020-04-18:
the problem with the ESP boards is true examples that could receive and send. for examples where it is separate there were no problems using only gdo0.

minkione, 2020-04-18:
Yeah, I noticed in the following examples:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/NewRemoteSwitch%20examples%20cc1101/NewRemoteRepeater/NewRemoteRepeater.ino
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/Rc-Switch%20examples%20cc1101/Repeater_cc1101/Repeater_cc1101.ino

LSatan, 2020-04-18:
maybe it could work now since a lot has been improved on the spi interface. I haven't tested it. I don't know if it would be wise to change it again if it is.

ps. i will change the thing we discussed yesterday if i find time. Thanks for your tip. ;)