LSatan/SmartRC-CC1101-Driver-Lib #30: setMHZ needs to change TEST0 register also

gusgorman402, 2020-05-03:
The setMHZ function needs to also change the TEST0 register from x09 to x0B for certain frequency ranges. The CC1101 frequency ranges become out of spec if the TEST0 isn't set correctly

https://e2e.ti.com/support/wireless-connectivity/other-wireless/f/667/t/88103?CC1110-at-400-MHz-with-example-sw-swrc088c

https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz/f/156/t/901879

LSatan, 2020-05-03:
Hi,
Many thanks for your message. To be honest, I've noticed that too. The data sheet hadn't helped me.
I will adjust it as soon as possible. 
Regards

gusgorman402, 2020-05-03:
Yea it was making me pull my hair out, luckily I found that TI forum thread. I simply added an IF THEN statement to the setMHz function
`if ((MHz >=322.88 && MHz <= 348) || (MHz >=430.5 && MHz <= 464) || (MHz >= 861 && MHz <= 928)){ SpiWriteReg(CC1101_TEST0, 0x09); }
else { SpiWriteReg(CC1101_TEST0, 0x0B); }`

LSatan, 2020-05-03:
I also assumed that it was due to my modules. did you test the settings with sdr? or do I have to look if the frequency ne has to be calibrated.
regards

gusgorman402, 2020-05-03:
It works. Fixed all my problems accessing the complete frequency range. The only issue I have now is about the fine calibration in the setPA function, changing the values of FSCTRL0. Where/why did you choose those values for FSCTRL0 when changing the PA for certain freq ranges?

LSatan, 2020-05-03:
ok nice to hear.
you ask why. the base frequencies were totally wrong at FSCTRL0 = 0. the settings are different for each frequency band. was tested with cc1101 as a transmitter in ask mode. and adjusted accordingly. tests were carried out for every frequency band with high medium and low settings. the problem was that i didn't get all the way down. that's why I asked if it fits.

gusgorman402, 2020-05-03:
I think FSCTRL0 settings will vary for different users & their chipset quality. I'll probably add a setOffset function to change the FSCTRL0 register. 
Besides that, your library is awesome. Best CC1101 library I've found so far. I'm building a cheaper version of the Yardstick One with it

LSatan, 2020-05-03:
Thank you for the flowers. The purpose of the library is to get everything out of the module. I am also of the assumption that it is different with different chip qualities. since i have different versions i will test it. offset function i think is a good problem that not everyone has sdr. but who has can;)

LSatan, 2020-05-04:
Hi I am currently writing the new information in lib. currently in the 433mhz band. That means setting the test0 register according to the specification. in addition, the frequency band is calibrated across the entire range when setting. since between 378 and 464 mhz there is a shift of 10.7 khz. Now I have discovered the following problem while trying out. if I go to 464mhz, test0 = 0x09 is sent as normal. now i go to 378mhz and it also sends with test0 = 0x0B. if I go back to 464mhz it is not possible. it is as if he had not set the test0 register to 0x09. funnily enough, it is set. if I now enter another value in test0 (example test0 = 0x0B) then it suddenly sends again. well it's funny but there is no other way. I will then test the other module as far as everything is. regads

LSatan, 2020-05-04:
problem solved
if it is done this way it fits.

`
if ((MHz >=322.88 && MHz <= 348) || (MHz >=430.5 && MHz <= 464) || (MHz >= 861 && MHz <= 928)){
 SpiWriteReg(CC1101_TEST0, 0x09);
byte s = ELECHOUSE_cc1101.SpiReadStatus(CC1101_FSCAL2);
SpiWriteReg(CC1101_FSCAL2, s+32);
 } else { SpiWriteReg(CC1101_TEST0, 0x0B); }
`
Regards

LSatan, 2020-05-04:
I once closed the cheap module from dsun. this module has a significant frequency shift. 300mhz + 36khz 340mhz + 41khz 378mhz + 45khz 464mhz + 55khz. everything based on my current calibration. I leave the 868mhz ban outside because it is a 433mhz module.

gusgorman402, 2020-05-04:
Sometimes the chipset is good quallity, but the crystal isn't
https://hackaday.com/2018/05/05/fail-of-the-week-never-assume-all-crystals-are-born-equal/