LSatan/SmartRC-CC1101-Driver-Lib #31: GDO0 & GDO2 on different pins

minkione, 2020-05-05:
Hi,
In Arduino UNO/NANO the GDO0 and GDO2 are allocated as follow:
D6 > GDO0 (TX)
D2> GDO2 (RX)

**Questions:**
```
1) Where I have to change (within the library/sketch) to be used with other Pins?
2) The D2 must be replaced with another INT pin?
```

Thanks a bunch for the reply!

LSatan, 2020-05-05:
hi,
do you use an ESP?
in the setup area with `ELECHOUSE_cc1101.setGDO(gdo0, gdo2);` Replace gdo0 with your pin. the same for gdo2. if a pin is not used set a 0.

LSatan, 2020-05-05:
ok i just see 0 is d3 at esp8266

minkione, 2020-05-05:
I was planning to use a Arduino UNO R3, but due to many reasons I have free for GDOs pins only D0, D1, D2, D3 

therefore I was thinking to use as:
GDO0 > D3
GDO1 > D2

Though, I wasn't sure where to declare these changes. 
Example if someone wants to use other pins than the suggested D6 and D2.

LSatan, 2020-05-05:
ok then so in the setup:
`ELECHOUSE_cc1101.setGDO(3, 2);`
Regards

LSatan, 2020-05-05:
which example do you use?

minkione, 2020-05-05:
Well, I was planning to use the samples types within these dirs:
- CC1101 default examples
- Rc-Switch examples cc1101

But before wiring the CC1101 was trying to figure out how to change the GDO0/GDO2 Pins or if they are needed at all for those sketches above.

LSatan, 2020-05-05:
ok I understand.
for the default examples setGd0 function. it is different for RC-switch.
with rcswitch: to receive ` mySwitch.enableReceive(2);`
to send `mySwitch.enableTransmit(3);`

minkione, 2020-05-05:
Ok gotcha, thanks man!
P.S. As far as I can infer... GDO2 must be an INT pin... instead the GDO0 no matter.
Correct analysis?

LSatan, 2020-05-05:
so, the gdo pins are not used for sending or receiving in the internal send / receive examples. the data for sending are written to the internal memory (fifo) via spi. then an instruction is given that the data is sent from the memory. when broadcasting, the cc1101 pulls up the gdo0 pin. when it is low again, the arduino knows that the transfer is complete. it is the same with reception. the arduino knows when the gdo0 pin is high and then low again that something has been received. here the cc1101 saves the received in fifo and thus that the gdo0 pin is high and then low again, the arduino knows that it can now read out the internal memory of the cc1101. 

The internal memory is not used for rcswitch and the gdo pins are configured so that when the cc1101 receives something, it is forwarded directly to the gdo pins. it is looped through directly so that the arduino can evaluate the signal. when the cc1101 is put into send mode it is the same as what goes into gdo0 is sent directly. So like the cheap receivers and transmitters. everything analog. by the way gdo0 can receive and send. gdo2 can only receive. if you hang a small loudspeaker between gdo0 and gnd you can hear the signal from your remote control.

 Conclusion. when the cc1101 is used like the cheap receiver. then specify (example rcswitch) where the send pin is connected and where the receive pin. rswitch then switches the corresponding arduino digital pin to receive input and send output. the cc1101 gdo pins are switched with setTx as input and setRx as output. it is nothing more.

 because what does set gdo do. it only sets the pins on input and output for aurdino. at rcswitch, rcswitch does it for the arduino.

minkione, 2020-05-05:
This is the best explanation ever seen in a Github issue!
You saved my time on reading line by line this awesome library!
Thanks for the explanation! :)

minkione, 2020-05-05:
Feel free to either close the issue or keep it open as reminder for others. :)

LSatan, 2020-05-05:
nice that I could bring you some light in the dark. I hope that everything is understandable. if some things sound strange then i apologize for my bad english. 

I would also like to add that setCCmode 1 configures the pins for the internal send / receive functions. if you set setCCmode back to 0 then the pins are configured for rcswitch and co in the same way. for rcswicht alone it does not have to be set since it is standard 0. but switching back and forth would also allow switching between rcswitch and internal send functions in one sketch.

minkione, 2020-05-05:
Forgot one thing... since I will have also to attach another SPI device on the same Arduino UNO + CC1101... since the D10 is used as SS_PIN... I guess I can change it in https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/ELECHOUSE_CC1101_SRC_DRV.cpp#L255

`SCK_PIN = 13; MISO_PIN = 12; MOSI_PIN = 11; SS_PIN = 10;`

With an unused Pin I have available. Example D0 or D1. Correct?

LSatan, 2020-05-05:
yes, you can easily create your own.

`ELECHOUSE_cc1101.setSpiPin (SCK, MISO, MOSI, CSN);`
Example:
`ELECHOUSE_cc1101.setSpiPin (13, 12, 11, 1);`

the 1 at the end for ss pin as d1. is also described in the readme. it is best to insert the whole thing before the init command. in the setup area

minkione, 2020-05-05:
Great, thanks man!

LSatan, 2020-05-05:
with pleasure. I am available for further questions. by the way, there will be an update tomorrow. improved frequency calculator, calibration of the frequencies. and bug fixes. if you need debug function you can use my debug tool to match the cc1101 lib. there are also advanced features like set data rate and so on.
https://github.com/LSatan/CC1101-Debug-Service-Tool

closed 2020-05-06 by minkione