LSatan/SmartRC-CC1101-Driver-Lib #33: Does it work for NodeMCU?

StefanoGiu, 2020-05-30:
Does it work for NodeMCU?

What is the wiring schema? Thanks!!! :)

LSatan, 2020-05-30:
Hi,
yes it works with nodemcu esp8266. use the wiring diagram of esp8266 wemos d1 mini.
[image]
Regards

StefanoGiu, 2020-05-30:
Thanks for the prompt reply!! No change in the code?

LSatan, 2020-05-30:
is not necessary. unless you want to use other pins.

LSatan, 2020-05-30:
I once read your other editions.
I guess it's about the Silentron plant. if you plan to use another mod for rc-swicht it will not work. raw data modulation in other types of modulation is not yet possible. but I'm working on it.

My guesses if your transmitter works with ask:

the problem will be that a rolling code is used. They are probably wondering why they receive nothing because you should still see something. the system probably uses the manchester coding. because it's very common. manchester cannot process rc-switch. Demodulating the Manchester signal is not a problem in itself. but you should first test what you have for a signal. then you can see what you're dealing with. use this example:
https://github.com/sui77/SimpleRcScanner/tree/master/htdocs/Sketchbook
Regards

StefanoGiu, 2020-05-30:
I used the MX-RM-5V and I was not able to read any raw signal... not even with the branch protocollessreceiver of Martin-Laclaustra and the ProtocolAnalyzeDemo.ino... no signal at all using the remote control or opening a window with the contact switch.

Then I decided to bough this one
https://www.amazon.it/gp/product/B07YX92NMP/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

and test this project - It will arrive on Monday.

I dediced to test this one cause, as I understand MX-RM-5V uses only ASK, while the last transreceiver can use also FSK, GFSK and MSK modulation. I don't know if Silentron uses ASK or not...

Do you think it will not work with it?

I'm a complete newby of those kind of thinks, and any help from your side to guide me through it is appreciated. I'd like to make my old Silentron alarm system --> SMART.

Thanks

LSatan, 2020-05-30:
it is difficult to say whether the system uses ask. I do not know it. if you used the sketch linked above and there was nothing, you should have seen 0.0.0.0.0.0.0.0.0, ..... in your edition. correct? I can only tell you so much that you will not receive anything with fsk and co because there are so many setting options for the module and it must be set exactly. what you can test if it sends on other frequencies with ask. if you want to find out what signal it is and on what frequency it is broadcasting you have to try sdr.
https://www.amazon.it/Portatile-Ricevente-Sintonizzatore-RTL2832U-Sostegno/dp/B013Q94CT6/ref=mp_s_a_1_8?dchild=1&keywords=sdr&qid=1590845930&sr=8-8

Here is some information about the software. But the article is in German. you can also use google and search "sdr with dvbt stick".
http://thomas-messmer.com/index.php/15-free-knowledge/allerlei/technische-informationen/6-sdr-via-dvb-t-stick

StefanoGiu, 2020-05-30:
If I buy the object you linked in Amazon, what shall I be able to achieve? It seems to be an antenna for DVB-T... isn't it?

LSatan, 2020-05-30:
You can use it to display radio signals acoustically and graphically, regardless of the modulation. between 0mhz and 1000mhz. You can use it to recognize the modulation and the frequency. You can also record and analyze the signal on audio. here is a sample picture.
https://www.google.com/amp/s/nl.oneguyoneblog.com/2016/09/13/sdrsharp-sdr-installeren-op-windows-10/amp/

StefanoGiu, 2020-05-30:
Looks promising... but even if I know the modulation and the frequency... then I wouldn't know how to use it with a receiver on ModeMCU and how to read my sensors... then if they implement rolling codes... it will be even more difficult. I would try and dig into it only if there is any chance to get it working and someone expert can guide me into it

LSatan, 2020-05-30:
it is best to wait until the cc1101 module is there and try 430mhz. with rc switch it can also be 870mhz i don't read the document because of the language.
https://www.yumpu.com/fr/document/view/30601766/silentron-italia-domuswire

StefanoGiu, 2020-05-30:
It says it transmits dual frequency 430 and 879 Mhz. I bought the antenna you suggested me. I'm also waiting for cc1101 to arrive and I already own MX-rm-5v... If you want and you have time to help me with it... I can do all the test you suggest me to run.

LSatan, 2020-05-30:
ok then we'll stay that way for now. if you have the module there, first wire it up and use the example receive demo simple cc1101 from my library. I think we should start at 430mhz. if you have done everything so far then contact me again.

StefanoGiu, 2020-05-30:
Thanks a lot for your time. I will contact back as soon as I will get the cc1101 module.

LSatan, 2020-07-31:
problem was solved successfully. communication via hangouts.