LSatan/SmartRC-CC1101-Driver-Lib #34: Repeater with CC1101 Examples

minkione, 2020-06-10:
Just got an idea of creating a repeater sketch (as existing in other examples directories) but with:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/CC1101%20default%20examples/cc1101_Receive/cc1101_Receive.ino

https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/CC1101%20default%20examples/cc1101_Transmitt_Hello_World/cc1101_Transmitt_Hello_World.ino

Is there any limitation that would block me?
I noticed that in the RX sketch the cc1101 is forced to RX:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/bd1cb184d825677bed52a832d91e09b25efb339a/examples/CC1101%20default%20examples/cc1101_Receive/cc1101_Receive.ino#L29

But in the TX sketch... it is NOT forced to TX (as mentioned in the available README):
`ELECHOUSE_cc1101.SetTx(); //Set transmit on.`

https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/README.md#instructions--description

Why?

LSatan, 2020-06-10:
Hi. I would not force it, it is only executed once in the setup. it is necessary for the cc1101 to go into receive mode. Settx is also set to send. this happens internally in lib in the send script. if you want to write a repeater you have to install setrx after the transmission.
Regards

minkione, 2020-06-10:
Ok gotcha.
Yeah the idea was to move the `ELECHOUSE_cc1101.SetRx();` in the loop() and use some IF statements to handle RX & TX asyncronously like in https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/bd1cb184d825677bed52a832d91e09b25efb339a/examples/Rc-Switch%20examples%20cc1101/Repeater_cc1101/Repeater_cc1101.ino#L49

LSatan, 2020-06-10:
Yes true

minkione, 2020-06-10:
thanks man!

closed 2020-06-10 by minkione