LSatan/SmartRC-CC1101-Driver-Lib #35: repeater_cc1101 seems to be working incorrectly

hacuongvv, 2020-06-14:
I want to make a device that repeats the RF signal for the garage door, the problem is that when I use the repeater_cc1101 example, the arduino can receive the code of the remote but when the playback is the signal is wrong, I used  1 cheaper RF receiver and see that the signal emitted from the cc1101 is jammed in and out of the RC-switch, can you help me fix it?  Sorry for my bad english!

LSatan, 2020-06-14:
Hi, you use an arduino. correct? what exactly is wrong with the values? a better description of what is wrong would be helpful. I will test the example again. but can only do it with my means.
Regards

LSatan, 2020-06-14:
is there a serial output at cc1101 repeater ino? if it has received something and then sends it, it outputs a serial output. if it is true, your cheaper receiver may not be able to receive the set frequency of the cc1101. some remote controls also transmit on 433.85 mhz. try changing the frequency and see if your cheap receiver receives it.

hacuongvv, 2020-06-14:
I used an oscilloscope to measure the signal at a cheap receiver, I used the arduino nano and arduino uno for the experiment, the results are the same, can't control the garage door, I'll show you the oscilloscope screen on tomorrow, my remote uses OOK

LSatan, 2020-06-14:
OK. since i'm not at home i can't do any tests right now. I'll check it out later.

LSatan, 2020-06-14:
I still have a question. does send demo cc1101 work properly?

hacuongvv, 2020-06-14:
I tried it and did not expect it, the signal from pin TX (PIN_6) of arduino is correct but cc1101 emitted incorrectly, it seems that the problem is happening here.

LSatan, 2020-06-14:
I try to get an exact picture of the problem. for me it always worked perfectly. for my receivers example garage door and also for radio-controlled sockets. I'm assuming your wiring is correct. only gdo0 can send. Both gdo0 and gdo2 can receive.

hacuongvv, 2020-06-14:
I connected the power cord as your guide, maybe I have to review it again, one more thing is that I see the maximum distance for c1101 to receive a signal from the remote is about 40m, I wonder if it is  limited yet

closed 2020-06-16 by hacuongvv