LSatan/SmartRC-CC1101-Driver-Lib #36: Does the order of functions matter after inititialization?

minkione, 2020-07-19:
Hi,
I am trying to debug some weird behaviour, but first a bit of overview:
I am working on a sketch that would support both rcswitch ASK/OOK modulation and also the classic CC1101 modulations... to switch from one to another type I use a int acting as boolean (see it as a button. If pressed, it will switch from RCswitch ASK/OOK to CC1101's 2FSK) All clear so far? :)

Now my question... After setting the SPIPins and Initializing every-time the CC1101 with _ELECHOUSE_cc1101.Init();_ ... are there any other functions that MUST be called in a specific order?

To me looks like... if I initialize the CC1101 with the:
```
 if (ClassicASK){
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.SetTx(freq);
  mySwitch.enableTransmit(pin);
  }
```
And then I try to switch to the other mode...
```
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.setGDO(gdo0,0);
  ELECHOUSE_cc1101.setCCMode(1); 
  ELECHOUSE_cc1101.setModulation(0); 
  ELECHOUSE_cc1101.SetTx(freq);
```
Something wrong will happen. I was looking in the library if there was a function to make a hardware reset of the CC1101 in ordfer to be sure that between a switch from one mode to another... I would be able to flush the CC1101's registers.

**Overall Idea of the sketch I am working on:**
```
int ClassicASK;
ClassicASK = 1;
<... BLABLABLA...>

void setup() {
 ELECHOUSE_cc1101.setSpiPin(52, 50, 51, 19);
<... BLABLABLA...>

}

void loop() {
<... BLABLABLA...>

 if (ClassicASK){
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.SetTx(freq);
  mySwitch.enableTransmit(pin);
  mySwitch.send(1337, 24);
  }
 else {
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.setGDO(gdo0,0);
  ELECHOUSE_cc1101.setCCMode(1); 
  ELECHOUSE_cc1101.setModulation(0); 
  ELECHOUSE_cc1101.SetTx(freq);
  ELECHOUSE_cc1101.SendData(transmitt_byte, 11);
  }

<... BLABLABLA...>

}
```

LSatan, 2020-07-19:
Hi,
it is enough to initialize the cc only once.

If you do it that way it should work.
`
bool ClassicASK = 1;
<... BLABLABLA...>

void setup() {
 ELECHOUSE_cc1101.setSpiPin(52, 50, 51, 19);
 ELECHOUSE_cc1101.Init();
<... BLABLABLA...>

}

void loop() {
<... BLABLABLA...>

 if (ClassicASK){
  ELECHOUSE_cc1101.setGDO(0,0);
  ELECHOUSE_cc1101.setCCMode(0); 
  ELECHOUSE_cc1101.setModulation(2); 
  ELECHOUSE_cc1101.SetTx(freq);
  mySwitch.enableTransmit(pin);
  mySwitch.send(1337, 24);
  mySwitch.disableTransmit();
}
 else {
  ELECHOUSE_cc1101.setGDO(gdo0,0);
  ELECHOUSE_cc1101.setCCMode(1); 
  ELECHOUSE_cc1101.setModulation(0); 
  ELECHOUSE_cc1101.setMHZ(freq);
  ELECHOUSE_cc1101.SendData(transmitt_byte, 11);
  }

<... BLABLABLA...>

}
`

Regards

minkione, 2020-07-19:
Unfrotunately, it was the first thing I have done.
Same strange behavior.

To me looks like, the CC1101 doesn't like switching from one mode to another within the void loop().
I also noticed that if I select the rc-switch ASK/OOK everything works as supposed... but as soon as I try to switch to CC1101's 2-SFK mode... the CC1101 starts TX a lot of noise and doesn't transmit the packet was supposed to.

I will keep digging and check closely library and CC1101 datasheet.
I have the feeling it something related to the fact it is switching from one mode to another within the same runtime.
Will keep you updated.

minkione, 2020-07-19:
Here a simple sketch that can replicate the issue:
P.S. Change SPI pins and transmitt/gdo0 accordingly to your board.

```
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>

int pin = 20; // int for Transmit pin.
int gdo0 = 20;

RCSwitch mySwitch = RCSwitch();

byte transmitt_byte[11] = {72,101,108,108,111,32,87,111,114,108,100};

void setup() {
  Serial.begin(9600);
  ELECHOUSE_cc1101.setSpiPin(52, 50, 51, 19);
  ELECHOUSE_cc1101.Init();  
  Serial.println("Finished Initialization"); 
}

void loop() {
    // Classic rc-switch ASK/OOK mode
    mySwitch.enableTransmit(pin);
    Serial.println("after enableTransmit");
    ELECHOUSE_cc1101.SetTx(433);
    Serial.println("after SetTx");
    mySwitch.send(31337, 24);
    Serial.println("after send ASK/OOK");
    delay(5000);
    //CC1101 2-FSK mode
    ELECHOUSE_cc1101.setGDO(gdo0,0);
    Serial.println("after setGDO");
    ELECHOUSE_cc1101.setCCMode(1);
    Serial.println("after setCCMode");
    ELECHOUSE_cc1101.setModulation(0);
    Serial.println("after setModulation");
    ELECHOUSE_cc1101.setMHZ(433);
    Serial.println("after setMHZ"); //### HERE WHERE IT GETS STUCK ###
    ELECHOUSE_cc1101.SendData(transmitt_byte, 11);
    Serial.println("after SendData");
}
```

When you will try it... you will see (with any SDR + URH or SDSharp or GQRX etc...) that the CC1101 does initialize and sends out the ASK/OOK packet...  but it get stuck right before sending the 2-FSK packet and it freezes the void loop(). It literally gets stuck.

LSatan, 2020-07-19:
you use the arduino mega. Correct?

minkione, 2020-07-19:
Yes. Currently I am using the MEGA with custom Pins for CSN and GDOs.
50 MISO
51 MOSI
52 SCK
21 GDO2
20 GDO0
19 CSN

If make sense... let me pick an ESP32 and see if I get similar errors.

minkione, 2020-07-19:
Tried with an ESP32 and another MEGA2560 with classic pins allocation (i.e. D53 for CSN and D6 for GDO0).
Let me know in the next days, when you will have time, if you can reproduce it. :)
I think I need to have a closer look at the CC1101 datasheet to read how it handles internal resets and registers.

minkione, 2020-07-19:
Just found this https://github.com/simonmonk/CC1101_arduino/issues/1

Tomorrow will try that

LSatan, 2020-07-19:
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>

int pin = 20; // int for Transmit pin.
int gdo0 = 20;

RCSwitch mySwitch = RCSwitch();

byte transmitt_byte[11] = {72,101,108,108,111,32,87,111,114,108,100};

void setup() {
  Serial.begin(9600);
  ELECHOUSE_cc1101.setSpiPin(52, 50, 51, 19);
  ELECHOUSE_cc1101.setGDO(gdo0,0);
  ELECHOUSE_cc1101.Init();  
  Serial.println("Finished Initialization"); 
}

void loop() {
    // Classic rc-switch ASK/OOK mode
    ELECHOUSE_cc1101.setCCMode(0);
    ELECHOUSE_cc1101.setModulation(2);
    
    mySwitch.enableTransmit(pin);
    ELECHOUSE_cc1101.SetTx(433);
    
    mySwitch.send(31337, 24);
    mySwitch.disableTransmit();
    Serial.println("after SendRcSwitch");
    ELECHOUSE_cc1101.SetRx();
    
    delay(5000);
    //CC1101 2-FSK mode

    pinMode(gdo0,INPUT);
    ELECHOUSE_cc1101.setCCMode(1);

    ELECHOUSE_cc1101.setModulation(0);
    ELECHOUSE_cc1101.setMHZ(433);

    ELECHOUSE_cc1101.SendData(transmitt_byte, 11);
    Serial.println("after SendData");
}

LSatan, 2020-07-19:
this works. the problem was that the gdo0 pin has to be set as input again. As a note: you can change the registers at any time and also in the loop

minkione, 2020-07-20:
Thanks.
May I ask the reason of this:
```
// Classic rc-switch ASK/OOK mode
ELECHOUSE_cc1101.setCCMode(0);
ELECHOUSE_cc1101.setModulation(2);
```
I didn't see being forced to 0 the CCMode in the repo's rc-switch examples.
Looks like I am missing some theory here. What's exactly is the CCMode for?

LSatan, 2020-07-20:
cc mode is an option that I added so that you can easily change several registers worth the cc1101. ccmode (1) switches the registers so that you can use the internal reception options. ccmode (0) switches the registers back so that you can use rc-switch. The library is set up in such a way that you can use rc-switch, so it does not have to be set with rc-switch. So if you want to change the registers for the internal reception options and then use the rc switch again, you must also reset the registers.

minkione, 2020-07-20:
Now is clear! I thought you were doing that to somehow reset registers! :)

Btw... regarding https://github.com/simonmonk/CC1101_arduino/issues/1 I was checking the datasheet and indeed says that after reset the register should old the value 0x29.

Maybe worth adding to your library Reset function?
Anyway, thanks again for the awesome work!

closed 2020-07-20 by minkione

LSatan, 2020-07-20:
I don't know why you want to reset the registers, but you already have ELECHOUSE_cc1101.setSres (); tries. the cc1101 has to be initialized afterwards.