LSatan/SmartRC-CC1101-Driver-Lib #37: Frequency Deviation and Bitrate

vekexasia, 2020-07-25:
Hello @LSatan ,

I found your library while debugging RadioLib for a project of mine. Apparently with ESP32 RadioLib struggles to work while yours works just fine. If you're curious you can find More on that here -> https://github.com/jgromes/RadioLib/issues/173

RadioLib supports a non blocking receive method (just like yours I guess?) But the problem is I need to also specify the bitrate and the frequency variant along with preamble length and syncword. I inspected your code and didnt find a way to actually specify that.

Is there a way to do that?

Thanks

LSatan, 2020-07-25:
Hello,
for further functions you can use the debug service tool. https://github.com/LSatan/CC1101-Debug-Service-Tool
. You can use the debug function to change any rigister during operation. or you only use the service function for devitan and so on. alternatively, each register can be used without the debug tool with 
ELECHOUSE_cc1101.SpiWriteReg (REGISTER, VALUE);  just add it in the setup.

you can find the register arrangement in the .h file

#define CC1101_SYNC1        0x04        // Sync word, high INT8U

ELECHOUSE_cc1101.SpiWriteReg (0x04, VALUE);

VALUE = your value in hex or dec

I am available for further questions.

Regards

LSatan, 2020-07-25:
description from the debug tool:

cc1101_debug.setDRate(100.00); //Set Data rate from min 0.02kBaud to max 1621.83kBaud directly like SmartRF Studio.

cc1101_debug.setDeviation(47.00); //Set Deviation from min 1.58khz to max 380.85khz directly like SmartRF Studio.

vekexasia, 2020-07-25:
hey @LSatan thanks for your reply. Let me try :)

LSatan, 2020-08-16:
many new settings available in the 2.5.0 update.
Regards

closed 2026-04-26 by github-actions[bot]