LSatan/SmartRC-CC1101-Driver-Lib #39: Cant stop transmitter after transmission

MACscr, 2020-07-29:
Ok, here is about as basic of an example as i can get, but while this should only transmit once, according to my SDR, it just keeps transmitting and a continuous signal is found on 303mhz until i unplug the arduino. What am i doing wrong?

```
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>

int pin; // int for Transmit pin.
String rfCode;

RCSwitch mySwitch = RCSwitch();

void setup() {
  
  Serial.begin(9600);
  
  #ifdef ESP32
  pin = 2; // for esp32! Transmit on GPIO pin 2.
  #elif ESP8266
  pin = 5; // for esp8266! Transmit on pin 5 = D1
  #else
  pin = 6; // for Arduino! Transmit on pin 6.
  #endif
  
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.setMHZ(303.869);
  ELECHOUSE_cc1101.setClb(1,13,14);
  
  // Optional set protocol (default is 1, will work for most outlets)
  mySwitch.setProtocol({ 372, { 35, 1 }, { 1, 2 }, { 2, 1 }, true });
  
  // Optional set pulse length.
  // mySwitch.setPulseLength(320);
  
  // Optional set number of transmission repetitions.
  mySwitch.setRepeatTransmit(1);

  // Transmitter on
  mySwitch.enableTransmit(pin);
  
  // cc1101 set Transmit on
  ELECHOUSE_cc1101.SetTx();

  rfCode = 123;
  
  Serial.println("rfCode sent to transmit = " + rfCode);
  mySwitch.send(rfCode.toInt(),12);
  delay(1000); 

  // disable transmitter
  mySwitch.disableTransmit();   // set Transmit off
}

void loop() {
}
```

LSatan, 2020-07-29:
Let me try

LSatan, 2020-07-29:
ok
add it:
ELECHOUSE_cc1101.SetRx();
after:
mySwitch.disableTransmit();

MACscr, 2020-07-29:
So it keeps transmitting something or even dead air until its turned back into receiver? That seems a bit crazy.

- Mark

LSatan, 2020-07-29:
so with settx the cc1101 is put into send mode. the difference between the cheap transmitters and the cc1101 is that the cheap transmitters still have a pull down resistance. the cc1101 doesn't have this. for this reason it is possible that some kind of light noise is sent. even if nothing is actually broadcast. an empty carrier. it is recommended that you put the cc1101 back into receive mode after sending. so that there is also peace. In my opinion, however, there is no disadvantage because you can switch between sending and receiving as often as you like.

LSatan, 2020-07-31:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/38#issuecomment-666056272

closed 2020-07-31 by MACscr