LSatan/SmartRC-CC1101-Driver-Lib #40: CC1101 module not initialising with an SD Card inserted (ESP32)

JohnJJG, 2020-08-04:
Hi,

Has anyone tried getting the cc1101 to work with an SD Card reader on ESP32.  In my case both are SPI, both work independently but the CC1101 will not initialise if an SD card is inserted.  They are both modules so include the pull-ups.  My guess is that the ESP32 SD card library is the problem but would be good to know if this

Configuration Info:
I am using OpenMQTTGateway (v0.94) with RF/CC1101 where I have also added in SD card capability.  Using an ESP32 Dev board from Espressif.

CC1101 Module:	https://www.aliexpress.com/item/32821911314.html
SD Card Module:	https://www.aliexpress.com/item/32886466598.html (not the one but identical)

MOSI=23, MOSO=19, SCK=18, CS_RF=5, CS_SD=33

LSatan, 2020-08-04:
Hi
I'm still on vacation until the weekend. So no data and esp32 there. I also connected an SD card slot to my esp32 with 2x CC1101. But it was a while ago.
as lib for sd i modified fat32 lib because she didn't want to work with esp32 (regardless of cc1101). I connected the SD module to default spi (csn = 5, sck = 18, miso = 19, mosi = 23).
The cc1101 to the second spi (csn = 15, sck = 14, miso = 12, mosi = 13).
I think I remember some SD Libs problems. that's why I used sdfat. if the owner has not yet done it, all entries (ifdef esp8266) have to be changed to (ifdef esp32) so that it works. If you haven't got it by the weekend, I'll take care of it.

JohnJJG, 2020-08-04:
Thanks @LSatan, really appreciate the quick response.  Using separate pins would probably avoid the issue, I am using just about every pin on ESP32 Pico so I might have to see if I can free 3 up. If that doesn't work i'll have a look at swapping out the library. Was hoping to use the same bus but this would be safer.

JohnJJG, 2020-08-05:
Thanks, managed to get it working by splitting into separate SPI.  Left OpenMqttGateway to handle the CC1101 since that only expects the receiver/emitter gpio pins and setup the SD card as below.  Got MISO on an input only pin and had to use the Rx (GPIO3) pin but seems to work. Thanks again.

SPIClass spiSD(HSPI);

void setupSD(void)
{
  pinMode(SD_CS, OUTPUT);
  Log.notice(F("Setup SD Card" CR));
  spiSD.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);  
  if(!SD.begin(SD_CS, spiSD)){
      Log.notice(F("Card Mount Failed" CR));
      return;
  }
   

CC1101:
MOSI=23, MISO=19, SCK=18, CS=5
SD:
MOSI=26, MISO=34, SCK=3, CS=33

LSatan, 2020-08-05:
I'm happy to hear that it worked.
Regards