LSatan/SmartRC-CC1101-Driver-Lib #41: Stuck at ELECHOUSE_cc1101.Init();

T-vK, 2020-08-13:
I can't get the cc1101_Transmit_Hello_World example to work. 
I added a few Serial.println() calls to pin the problem down.
As you can see, it never makes it past the `ELECHOUSE_cc1101.Init();` method.

[image]

My hardware: 

- [Arduino UNO clone](https://www.aliexpress.com/item/32556087234.html)
- [D-Sun CC1101 Module](https://www.aliexpress.com/item/32824627014.html)
  [image]

My wiring:

```
GDO2 -> -
GDO0 -> 6
CSN -> 10
MOSI -> 11
MISO -> 12
SCK -> 13
GND -> GND
VCC -> 3v
```

Any ideas what is going wrong?

LSatan, 2020-08-13:
Hi, i will test it quickly. and try to reproduce the error.
Regards

T-vK, 2020-08-13:
I just grabbed another one of these Arduino clones and a new CC1101 module and I still had the same issue.

Then I used a different cable and with that it's actually working. ??? I'm really confused because uploading the code worked consistently and it also consistently stopped at the init() method. 
Never in a million years would I have guessed that a faulty USB cable could behave like this.
Sorry for bothering you with this...

LSatan, 2020-08-13:
no problem. I was just testing it and couldn't find any problems. nice that it works now.

T-vK, 2020-08-13:
Thanks for the quick response btw.

closed 2020-08-13 by T-vK

pascal260303, 2021-11-28:
I had the same issue and also fixed it with changing the cable. I don't understand it

pochovavac, 2021-12-01:
I also have this problem and it was a temporary solution for me
1. Disconnect the USB cable
2. press and hold the reset button on the Arduino
3. connect the USB cable
4. open the serial monitor
5. Release the reset button

chrisonair, 2024-07-05:
I get the same thing. Tried 4 different cables, different USB ports. I cant get passed the initialization.. spi says connection ok. Then i do the init and it hangs there