LSatan/SmartRC-CC1101-Driver-Lib #44: Send and receive more than 32 bits

joelsernamoreno, 2020-08-25:
Hello!

I would like to send and receive more than 32 bits

Is this possible?

Example:
CC1101_1.send("11100010110001000101110001001000100001000101001001000");

This is the signal from a weather station

Thank you!

LSatan, 2020-08-25:
Hello,
If you want to use the internal sending or receiving methods, it is a little difficult without detailed knowledge. the cc1101 has an infinite packet length mode. however, the baud rate and other factors also play a role so that you can receive and send. take a look at this example: https://github.com/gusgorman402/RFmoggy/blob/master/rfmoggy_wifiClientMode.ino

In the simple mode it is a little easier to look at these examples:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master/examples/RemoteSensor%20examples%20cc1101
or this:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master/examples/ESPiLight%20examples%20cc1101

in short yes it is possible with the internal examples. and in the simple mode it works all the more.
Regards

joelsernamoreno, 2020-08-26:
Hello!

The examples you told me do not detect the signal

Well, this signal is simple, it's OOK without encryption

Here you can see what is being sent, taking into account that the longest pulses is 1 binary and the short pulses is 0 binary:
111000101100010001011100010010001000010001010011110

[image]

Using an SDR I can see that the temperature is sent

[image]

Well, I would like to be able to receive and send more than 32 bits with the Send function, in order to make a replay attack

Attached are screenshots of the signal

Thanks!

LSatan, 2020-08-29:
I'll take a closer look at it. so please be patient.
Regards

joelsernamoreno, 2020-08-29:
Hello!

Don't worry, I'm in no hurry for this

Well, I found a PR from the rc-switch library where a person implemented this, but the PR was never accepted

https://github.com/sui77/rc-switch/pull/221

I will try to implement this in your library, but I think you know better about your code and you can do better

Thank you!

LSatan, 2020-08-29:
the problem is not sending 64 bit. we can also send or receive 1000bit. the signal must be properly decoded. if I convert your bit sequence into decimal, nothing useful comes out of it. now i have found the following about the hideki protocol.
http://members.upc.nl/m.beukelaar/Crestaprotocol.pdf

 I think you can build on that. it would be good to know how many microseconds the jleonst pulse has.

LSatan, 2020-08-29:
can you make an audio recording of the signal and then upload it? the current temperature would be helpful.

joelsernamoreno, 2020-08-29:
Hello!

Yes, here is the I/Q file:
https://mega.nz/file/Q5ZVyQCT#Lgg0iHQtVzOVScShykjUx5OQ5doDGmgoejHGcAXfz80

Import this into an audio editor (example: audacity) with the following settings:

Encoding: 32-bit float
Sampling frequency: 2000000 Hz

joelsernamoreno, 2020-08-29:
Supposedly the temperature is 31.9ºC

joelsernamoreno, 2020-09-16:
Hello!

Any update?

Thanks!

LSatan, 2020-09-16:
hi, unfortunately i don't have time at the moment because i am doing advanced training and exams are pending. the signal of the raw data you sent me are unfortunately very weak. it would be good if you could record raw data with the help of the cc1101. i will post the sketch for it in the course of the day.
Regards

LSatan, 2020-09-16:
this is a prototype of my raw data scanner. there is also a built-in frequency scanner for remote controls. he won't bring you anything at the weather station. set the correct receiving pin (esp or arduino pin). if you do not receive anything try to set the bandwidth from 58khz to 812khz. the count_threshold is set to min 50 pulses. because weather stations usually send less than 100 pulses. You can play around with the settings if it doesn't work. monitor with sdr whether it is also the weather station that you have received. then send me the raw data. if it doesn't work or if another problem occurs, please contact me.

``` C++

#include <ELECHOUSE_CC1101_SRC_DRV.h>
#define SAMPLESIZE 500

int RxPin = 2; //receive pin
float freq = 433.92;
float start_freq = 433;
float stop_freq = 434;
int rssi_threshold = -70;
int count_threshold = 50;
int time_threshold = 70;

static int interruptPin = 0;
int state = 0;
long compare_freq;
float mark_freq;
int rssi;
int mark_rssi=-100;
int counter = 0;
static unsigned long lastTime = 0;
static unsigned long resetTime;
bool Receive = 0;
int MODE = 2;
String Print;
static unsigned int timings[SAMPLESIZE];

void setup() {
Serial.begin(115200);
pinMode(RxPin,INPUT);
interruptPin = digitalPinToInterrupt(RxPin);
ELECHOUSE_cc1101.Init();
ELECHOUSE_cc1101.setRxBW(58);
ELECHOUSE_cc1101.setMHZ(freq);
ELECHOUSE_cc1101.SetRx();
Serial.println("Raw data Receiver Ready!");
Serial.print("Freq: ");
Serial.println(freq);
}

void loop() {

commands();
print_received();
scan_frquency();

}
/////////////////////////////////////////////////////////////////////////////////////////////////
void commands(){
if (Serial.available()){
String c = Serial.readString();
if (c=="scan\n"){
MODE = 1;
}
}
}
/////////////////////////////////////////////////////////////////////////////////////////////////
void scan_frquency(){
if (MODE == 1){
serial_print("---------------------\nFrequency scan Mode\n---------------------");
set_state(2);
ELECHOUSE_cc1101.setMHZ(freq);
rssi = ELECHOUSE_cc1101.getRssi();

if (rssi>rssi_threshold){
if (rssi > mark_rssi){
mark_rssi = rssi;  
mark_freq = freq;
}
}

freq+=0.01;

if (freq>stop_freq){
freq=start_freq;

if (mark_rssi>rssi_threshold){

long fr = mark_freq*100;

if (fr == compare_freq){
Serial.print("Freq: ");
Serial.println(mark_freq);
Serial.print("Rssi: ");
Serial.println(mark_rssi);
ELECHOUSE_cc1101.setMHZ(mark_freq);
mark_rssi=-100;
compare_freq = 0;
mark_freq = 0;
MODE = 2;
delay(1000);
}else{
compare_freq = mark_freq*100;
freq = mark_freq -0.10;
mark_freq=0;
mark_rssi=-100;
}
}
}
}
}
/////////////////////////////////////////////////////////////////////////////////////////////////
void print_received(){
if (MODE == 2){
serial_print("---------------------\nReceive Mode\n---------------------");
set_state(1);
if (Receive == 1){
Serial.print("Counter: ");
Serial.println(counter);
for (unsigned int i = 1; i< counter; i++) {
Serial.print( timings[i] );
Serial.print(",");
}
Serial.println();
delay(500);
Receive = 0;
counter = 0;
set_state(1);
}
}
}
/////////////////////////////////////////////////////////////////////////////////////////////////
void serial_print(String s){
if (s != Print){
Print = s;
Serial.println(s);
}
}
/////////////////////////////////////////////////////////////////////////////////////////////////
void set_state(int s){
if (s == 1 && s != state){
state = s;
attachInterrupt(interruptPin, handleInterrupt, CHANGE);
}
if (s == 2 && s != state){
state = s;
detachInterrupt(interruptPin);
}
}
/////////////////////////////////////////////////////////////////////////////////////////////////
void handleInterrupt() {
static unsigned long lastTime = 0;
const long time = micros();
const unsigned int duration = time - lastTime;

if (millis()-resetTime > time_threshold){
if (counter > count_threshold){
Receive = 1;
set_state(2);
}else{
counter=0;
Receive = 0; 
}
}

if (duration > 100){
timings[counter] = duration;
counter++;
}

if (counter >= SAMPLESIZE){
Receive = 1;
set_state(2);
}

lastTime = time;
resetTime = millis();
}

```

joelsernamoreno, 2020-12-28:
Hello!

I have tested your code, I receive the following:

Rebooting...
⸮⸮⸮⸮`⸮C⸮3⸮Q⸮⸮⸮⸮⸮⸮⸮j⸮kC;ʱ⸮qpE⸮;S⸮0⸮}f⸮⸮0*⸮⸮⸮:=⸮⸮;⸮s⸮9⸮p9⸮⸮c02⸮8yCvp`q02⸮⸮8yCv`⸮0Y|⸮⸮2⸮Raw data Receiver Ready!
Freq: 433.92
---------------------
Receive Mode
---------------------
Counter: 500
339,8459,987,340,326,968,989,325,333,969,315,988,984,332,318,980,333,970,969,320,336,968,995,315,334,971,319,991,965,335,326, 971,344,978,982,322,324,960,1000,320,335,964,324,971,339,980,320,975,332,978,321,1301,353,8455,975,325,316,982,984,324,338,964, 330,973,971,337,332,972,325,984,982,323,326,983,955,345,337,961,324,974,979,339,316,966,340,986,979,324,326,958,1002,319,318, 985,328,971,324,984,332,970,325,962,325,1298,335,8449,986,349,297,987,981,325,316,984,333,992,969,323,332,972,319,984,986,326,321, 963,987,324,340,962,331,973,994,327,330,971,319,989,960,326,351,960,981,325,324,989,325,966,326,985,328,965,323,985,332,1294, 335,8476,986,331,314,985,957,353,314,982,332,973,972,325,331,968,320,986,987,333,319,962,984,327,324,985,329,973,990,313,351, 964,315,986,978,325,325,960,1002,323,327,958,353,969,337,961,323,974,337,957,345,1299,327,8478,986,317,340,960,981,326,327,982, 328,974,972,340,309,989,317,985,986,327,321,981,959,328,338,964,332,993,971,324,333,968,319,986,986,333,317,963,982,324,342,958, 359,970,296,984,333,991,320,963,332,1318,338,8455,988,331,312,984,985,324,317,981,334,971,969,323,334,974,317,982,989,330,322, 986,987,300,325,986,327,973,974,337,332,971,316,982,982,327,334,965,997,324,307,986,324,973,339,982,319,974,343,957,323,1298, 334,8474,985,324,325,981,983,328,334,965,321,973,1003,309,323,996,315,985,983,317,327,985,980,324,324,956,351,973,976,327,325, 971,343,956,984,320,340,964,987,320,323,985,333,967,324,985,330,970,338,966,321,1303,356,8463,979,324,328,982,979,320,325,985,324, 974,975,327,327,971,323,984,981,324,316,987,981,322,328,982,329,974,966,335,325,973,336,983,973,323,338,958,994,318,338,958,345, 976,337,955,325,970,342,959,350,1296,335,8462,982,324,329,980,982,324,314,987,328,970,975,330,327,972,339,965,998,323,337,963, 979,328,334,961,327,969,993,313,345,976,335,948,995,320,339,959,1000,319,317,983,329,968,315,986,331,973,343,964,329,1296,334, 8466,985,325,325,981,984,324,324,982,328,973,971,329,328,968,342,961,1010,305,323,982,981,323,327,983,330,971,977,339,326,970,
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x4000127a PS : 0x00060330 A0 : 0x800d3114 A1 : 0x3ffb1ef0  
A2 : 0x3ffb853c A3 : 0x000003ca A4 : 0x3ffc0d00 A5 : 0x2bda9294  
A6 : 0x000000 A7 : 0x000000 A8 : 0x0000002d A9 : 0x3ffb1eb0  
A10 : 0x3ffb1f60 A11 : 0x000000 A12 : 0x0000003c A13 : 0x3ffb857c  
A14 : 0x00060320 A15 : 0x00000003 SAR : 0x0000001c EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000003ca LBEG : 0x4000c46c LEND : 0x4000c477 LCOUNT : 0x000000  

Backtrace: 0x4000127a:0x3ffb1ef0 0x400d3111:0x3ffb1f00 0x400d315a:0x3ffb1f20 0x400d0dd4:0x3ffb1f40 0x400d0fe6:0x3ffb1f60 0x400d11ba:0x3ffb1f90 0x400d31fd:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Rebooting...
⸮⸮⸮⸮`⸮C⸮3⸮Q⸮⸮⸮⸮⸮⸮⸮j⸮kC;ʱ⸮qpE⸮;S⸮0⸮}f⸮⸮0⸮0*⸮⸮⸮:=⸮⸮;⸮s⸮9⸮p9⸮⸮c02⸮8yCvp`q02⸮⸮8yCv`⸮0Y|⸮⸮2⸮Raw data Receiver Ready!
Freq: 433.92
---------------------
Receive Mode
---------------------

Is this right?

Thanks

LSatan, 2020-12-28:
yes I can do something with that. The Corepanik error probably comes from the fact that it is better to write the data into the RAM with an esp. but you received something. give me some time to decode it.

joelsernamoreno, 2020-12-28:
Hi

Well, I modified your code to get the binary code

But... this releases many codes, how can I optimize your code to receive only the necessary code?

This is ASK, if I try to receive the code with a basic example (Evil Crow RF repo) I can't receive it because the harmonics are too big

But if I try to transmit the code, the device turns on

The code is this: 101001001010010100000

I think that the cc1101 modules cannot receive this because of the harmonics, but if I transmit the code my device turns on. My device is a wireless calling system

Thank you!

LSatan, 2020-12-28:
now I was also just fetig to decode. it is a 24 bit code.
my results: 101001001010010010100000
as I mentioned already rc switch does not comeclear with everything. It's not a problem for me to write a send and receive sketch.

LSatan, 2020-12-28:
I think rc-switch receives itnot because there is a stop signal.

joelsernamoreno, 2020-12-28:
Can you modify this example to receive the decoded code directly?

This way I can implement it in Evil Crow RF and modify the RSSI, Count and Time parameters from the web panel

LSatan, 2020-12-28:
yes this is the plan, but it requires a comprehensive analysis of the received raw data and very complex. so that everything is covered.

Writing something for your remote control is a bit easier. can then only receive signals that they are like those of their remote control.

cgmckeever, 2020-12-28:
I may be out of line here, but there is a brnach of `rc-switch` that does just allow the sending of the RAW data. I think the branch is a bit behind current, but the developer is intending to get it in. Not sure if this helps you at all

gkhnbks, 2022-01-28:
Hello, LSatan,

I have a ask remote control from china,
with the above code ı receive this codes, how can decode this remote and use in my code?

- [ ]  `  ---------------------

10:50:52.082 -> Frequency scan Mode
10:50:52.082 -> ---------------------
10:50:53.653 -> Freq: 433.99
10:50:53.653 -> Rssi: -60
10:50:54.653 -> ---------------------
10:50:54.653 -> Receive Mode
10:50:54.653 -> ---------------------
10:50:57.022 -> Counter: 295
10:50:57.022 -> 1408,624,393,645,370,644,379,650,384,629,612,409,613,409,612,409,610,411,612,413,384,631,387,655,610,389,640,384,637,387,384,632,635,412,386,636,611,411,388,632,609,410,612,408,387,640,611,411,386,633,610,409,614,410,382,638,387,636,388,633,387,633,387,633,391,634,614,414,387,633,608,411,385,633,389,636,387,632,387,633,385,635,615,413,615,409,385,636,608,410,383,636,388,637,389,637,383,12820,1413,632,382,633,390,636,385,637,387,634,610,409,609,414,613,413,611,409,610,408,385,634,387,634,615,413,611,415,609,410,384,636,608,409,389,636,614,414,387,633,609,408,609,414,388,636,613,409,386,634,610,414,610,412,386,635,387,637,384,635,389,636,387,636,385,634,610,408,389,636,614,411,386,637,383,635,385,634,386,635,388,639,615,413,611,409,384,635,610,410,386,637,388,635,385,634,385,12821,1416,634,386,636,385,632,386,634,387,636,614,408,614,408,611,408,608,412,614,414,387,634,385,634,610,408,615,412,618,411,385,634,611,408,388,638,615,408,385,635,610,412,615,409,388,634,614,410,388,630,611,415,612,413,389,635,383,637,386,633,389,633,387,632,386,632,616,413,388,637,610,408,384,634,389,636,389,634,389,634,386,632,610,409,614,415,387,635,615,410,383,633,387,635,386,637,389,1312862,1390,645,377,623,407,628,384,630,385,632,638,387,613,410,635,385,635,410,612,388,388,631,412,608,636,410,611,385,637,412,389,607,635,410,385,634,613,413,385,633,610,410,616,412,386,634,610,410,387,631,610,410,613,413,389,634,386,632,386,634,388,636,388,637,387,631,610,409,388,635,614,413,388,635,385,633,386,635,388,636,388,635,610,409,610,413,390,632,614,413,386,633,385,633,388,636,388,12830,1384,655,359,657,387,634,388,633,388,633,610,409,609,412,614,413,613,410,609,409,384,635,387,636,614,409,609,410,609,408,388,637,614,409,387,636,610,408,384,634,615,413,614,409,385,633,609,414,388,637,613,409,609,409,384,634,388,638,388,637,384,635,384,634,388,638,615,413,385,634,609,408,389,637,388,636,384,634,385,634,383,635,640,388,615,413,384,634,609,408,388,638,389,636,389,633,385,12818,1411,634,383,635,388,637,387,635,383,
10:51:01.272 -> Counter: 295
10:51:01.272 -> 1407,633,390,641,372,653,372,644,394,619,626,405,629,382,632,410,608,410,609,410,384,630,386,631,635,412,610,409,610,409,386,633,611,409,387,636,611,413,388,632,610,408,609,408,409,613,639,391,382,635,634,390,632,408,383,636,389,631,389,623,391,646,369,644,378,651,386,626,612,409,384,632,409,632,386,634,387,635,384,636,384,634,385,636,383,637,388,638,609,415,382,637,382,637,382,637,382,12837,1391,656,361,656,388,631,383,637,384,637,608,417,606,413,610,409,634,383,634,415,361,638,384,635,633,406,611,411,615,414,388,615,634,406,381,635,615,400,386,640,616,397,623,393,393,644,622,403,379,649,608,410,608,412,384,631,385,630,412,611,409,609,413,632,386,632,387,634,613,416,385,634,385,634,386,634,388,637,387,636,384,635,384,635,385,636,385,639,611,412,386,634,386,634,385,634,385,12844,1396,633,385,632,389,634,387,637,384,636,609,410,631,382,635,412,617,387,616,407,383,635,411,632,608,415,609,411,616,387,408,616,631,408,379,634,615,396,396,649,618,394,624,403,378,652,608,408,383,630,635,386,636,390,383,631,410,633,386,633,384,634,387,636,384,635,384,636,610,409,383,636,407,634,385,635,385,636,382,635,383,637,381,637,405,614,388,635,616,403,400,640,374,644,395,615,396,1148335,1395,645,371,645,378,649,384,628,385,630,635,409,611,410,613,412,613,386,638,385,410,633,384,635,616,411,613,387,636,385,409,633,611,410,385,634,610,413,389,634,610,409,611,413,387,636,614,412,386,632,610,410,610,409,385,636,389,637,388,635,385,635,385,635,386,635,389,637,611,409,385,635,384,635,388,638,384,637,385,637,384,634,409,610,408,613,410,636,615,382,407,635,389,614,402,641,374,12838,1383,645,394,619,401,623,405,628,383,630,610,412,635,387,635,385,634,410,610,412,385,632,385,634,610,413,610,410,609,409,387,635,614,413,385,635,610,409,384,634,634,407,608,414,381,638,607,404,388,639,606,402,629,399,396,649,368,644,372,645,379,651,383,628,385,631,387,657,610,410,384,635,384,633,385,634,384,636,384,636,384,637,384,636,384,636,408,637,609,404,389,632,373,655,366,640,395,12815,1420,622,380,652,383,630,385,632,412,
10:51:05.072 -> Counter: 295
10:51:05.072 -> 1394,645,372,647,378,652,381,629,381,630,636,386,634,408,611,408,613,409,615,414,387,632,387,631,611,409,615,414,611,413,387,632,610,409,388,635,614,412,386,633,610,410,615,411,388,636,608,411,384,635,613,415,613,410,386,632,386,635,387,638,389,634,387,630,385,637,613,413,615,409,386,633,386,638,384,637,388,636,385,634,385,635,614,415,389,632,386,632,611,409,388,639,386,637,384,635,385,12830,1389,656,360,658,384,635,383,631,384,639,610,413,616,408,611,409,607,415,613,409,391,634,389,636,612,407,613,407,613,406,388,637,615,409,390,632,611,410,388,631,617,411,614,412,388,631,613,408,389,633,614,413,616,410,383,633,385,635,389,635,387,638,383,635,385,635,613,415,613,410,385,633,390,636,389,632,387,633,386,633,386,634,612,413,385,639,385,636,610,408,385,639,385,636,388,636,388,12828,1411,630,386,632,386,633,387,635,387,635,610,409,610,407,616,413,608,413,613,411,386,635,387,632,610,408,639,382,635,389,412,612,635 [... cut ...] 384,637,634,384,407,634,606,404,389,629,391,646,392,618,399,12816,1403,649,356,653,384,630,387,632,386,632,635,410,609,409,612,412,613,412,610,409,384,633,385,634,609,414,631,388,631,397,390,625,618,419,370,645,603,407,407,628,609,411,612,409,384,631,635,386,386,659,608,387,638,385,410,632,386,634,386,633,385,636,385,634,388,636,386,634,385,637,610,408,384,634,385,635,408,613,408,638,383,635,381,638,606,414,380,638,615,403,403,634,374,630,397,623,391,12827,1400,645,378,650,382,629,384,631,385,
`

mtojek, 2022-07-01:
Hi @LSatan,

Great library port! Would you mind sharing some hints on how to decode timings into ASK? I have Hormann gate's remote and would like to decode the signal:

```

Freq: 433.92
---------------------
Receive Mode
---------------------
Counter: 126
2256,736,728,720,736,704,9960,1096,9560,1120,720,1104,1464,7640,1512,1104,704,1464,7632,10336,1496,1096,728,1088,1088,3424,4952,1104,1104,1096,1080,5712,736,15752,256,104,648,128,664,400,272,104,2224,712,720,704,712,728,4552,1080,3032,1488,1088,712,1104,1064,104,3768,1880,1080,1480,1088,1088,1056,5720,1112,720,1480,712,8024,1104,1112,8416,1112,752,1088,1112,1104,1456,720,16096,15760,248,712,1512,728,728,728,9576,720,1104,1088,1472,5712,2280,1112,728,1096,1480,1096,1064,1072,712,1056,3784,1080,704,1472,384,232,5712,1120,1088,1112,1104,8424,1104,712,1120,1088,1072,6104,1112,1112,53272,696,688,1488,728,736,720,720,704,720,704,4152,1072,2648,736,1504,1104,10328,9952,1456,11104,1128,1096,1104,1496,1088,1072,1080,696,1480,1064,104,344,328,14176,1112,1072,696,1472,360,640,1104,696,104,16056,736,744,1488,720,720,704,704,6480,3824,736,1480,1088,1096,696,1104,1056,1056,104,752,648,672,104,760,640,104,1072,352,656,1072,2248,736,1112,1088,1464,696,5720,1504,704,1104,8424,1112,1128,1120,704,8408,2264,1104,1096,1072,8016,1096,1096,728,16072,704,704,5352,720,728,4168,1072,1080,2264,1504,1104,1120,712,1088,1096,9576,1104,704,1096,1080,1480,696,1456,680,1112,1496,736,2248,1104,1080,1096,1064,104,6080,1120,1088,9944,1512,1080,1088,1120,1080,1080,4192,1488,1104,1096,
Counter: 73
368,256,104,624,128,2264,728,10736,728,1104,1432,104,696,1064,6880,1112,1112,1088,1464,664,1072,6096,728,1472,1104,1072,672,104,1072,1064,1072,1048,104,768,248,128,1072,1104,1120,1104,1472,1080,1056,1096,680,1472,1112,1864,1096,1112,1480,680,1464,392,624,1040,1112,1120,1120,1120,1096,1088,704,1440,384,624,104,1048,104,376,216,42112,280,104,672,1512,736,728,696,720,704,712,680,680,4536,368,216,1464,1072,1856,1504,1128,696,1496,704,1072,1080,760,656,1048,104,2224,736,11872,728,1096,1088,1472,712,1072,104,1080,1072,760,248,104,1040,1096,1504,1104,1112,1104,680,784,648,1088,104,5680,728,1480,1096,8024,744,1480,1096,1064,680,
Counter: 61
240,104,368,272,344,136,152,120,368,176,144,360,256,120,712,736,696,712,712,728,8416,1120,1480,704,1080,1464,680,3784,728,1120,1488,1088,9192,1104,704,1472,1072,1080,7632,1088,1072,1496,1072,368,640,400,632,104,5320,1104,1112,1480,5736,2248,1128,1096,1088,704,1472,62952,688,688,656,104,696,344,256,3800,744,4536,704,1448,104,4160,1112,1088,1456,680,1080,1080,4560,1104,1488,704,1456,6880,1104,1096,1096,712,1472,704,5744,1096,1512,1096,1096,1080,704,104,1440,4184,1096,1464,696,1464,1056,1072,1056,368,632,1056,104,384,624,104,1064,1072,688,3800,
Counter: 129
128,176,3792,736,704,720,696,680,664,3832,576,104,384,424,200,360,104,192,728,216,3800,1104,1096,9944,1488,1104,728,9592,3008,1104,1080,1064,1072,4936,2256,1112,704,1432,1072,5736,712,1472,1104,1088,1064,400,608,888,104,1032,120,368,464,120,104,3376,1104,1128,16544,224,680,392,224,2264,728,728,704,728,704,712,3856,608,392,104,424,104,648,128,744,200,128,752,200,424,104,2240,2272,1088,1096,680,1080,7248,1128,696,1432,104,704,776,232,120,3776,2272,1104,1088,10368,1088,1088,1112,704,776,632,1056,696,104,752,232,4552,1472,1080,1112,1104,1072,1080,4184,1496,34144,112,144,128,360,200,104,384,240,104,2248,752,728,728,704,672,704,7640,1496,728,1456,104,1064,1080,6864,1112,704,1472,7640,1520,1104,9936,1120,1096,1072,1456,1064,1056,6480,1120,1096,1456,680,6104,1120,1128,1064,1088,400,248,104,20352,104,184,120,368,248,104,3000,736,728,744,704,712,696,4528,392,232,128,1032,368,680,3792,2272,1088,1072,104,5712,720,1504,704,1096,1072,1072,1088,5704,2240,1472,104,672,1080,1056,104,1080,744,128,160,120,1024,128,1056,104,752,256,3792,1120,720,1488,1088,720,1072,104,1040,1024,4920,1088,1096,1088,8000,1080,712,
Counter: 242
736,608,120,512,176,240,1104,1120,728,1112,800,632,696,104,1040,1432,104,384,272,792,656,368,200,152,872,112,104,1056,2240,2248,1496,688,104,1064,1456,384,104,192,104,4920,2264,720,9536,1104,1848,1480,1104,704,8392,1104,1088,392,624,1072,6880,1128,712,15752,104,192,120,2240,1496,712,736,688,720,720,672,4128,120,3776,2264,704,784,280,1456,696,1072,760,608,128,392,224,128,736,648,688,768,216,1080,776,616,360,224,128,1408,392,240,104,5680,1072,1448,1088,1080,368,616,128,376,656,4184,1096,1104,784,256,1456,1072,6496,1112,1104,1056,1480,688,104,8376,1096,1056,104,376,232,128,1448,1056,384,592,376,256,39984,216,120,2248,1512,736,680,728,9944,1104,1104,1488,696,1072,1080,7256,696,1456,688,1432,688,6872,1480,9952,1120,1056,1472,1096,1080,1064,5304,3776,704,1480,1088,1056,384,624,3416,1104,1120,728,1344,1096,1072,16520,224,104,2240,1504,720,704,696,12224,1496,712,1112,1056,128,7232,720,1488,672,1472,696,1072,776,248,1408,128,5304,1080,1480,696,1080,760,264,760,248,752,256,6096,1096,1056,6488,2640,704,1448,1064,1040,104,1080,4576,1104,688,1352,392,616,120,1040,104,18688,120,128,120,1488,1512,728,752,712,712,704,712,720,4112,2640,728,2264,800,248,104,1048,1440,400,232,3808,2248,1088,1080,1088,1024,128,384,272,768,624,912,128,4176,1480,1072,1064,1056,104,376,232,5720,1120,1096,1056,1360,104,1072,104,5688,728,1104,1104,1072,1472,672,104,1424,128,904,104,360,288,128,272,376,120,448,368,104,208,120,248,104,2256,1120,1120,704,1464,18848,128,2248,1496,736,712,704,680,392,248,680,104,3832,224,128,3376,1472,712,1064,784,208,1456,104,648,128,7216,1056,1088,1048,120,656,104,768,632,392,528,104,5688,1088,1088,1056,680,5680,2264,103 [... cut ...] 712,1464,18840,128,360,120,192,368,144,152,368,128,104,192,472,128,104,360,120,128,680,736,728,720,720,696,4536,104,672,104,736,184,104,104,152,128,3376,688,776,640,680,1064,1416,128,4152,1496,1104,744,1464,1080,384,592,128,5680,704,1104,1064,104,1456,704,1424,120,1848,3424,1104,1096,1456,1064,368,640,104,368,632,680,104,2640,1872,1104,1080,1488,696,1432,104,384,600,120,1088,2272,2256,1088,1096,688,104,1440,18816,120,216,1480,720,728,704,712,736,696,9168,744,1104,1456,8016,728,1112,1496,501,
```

mtojek, 2022-07-01:
I figured it out, so let me leave a hint for the following researchers. The signal I posted is Manchester encoded and it's proprietary. You won't be able to decode it using standard lib API.