LSatan/SmartRC-CC1101-Driver-Lib #45: Problem extending library through inheritence

coylen, 2020-08-26:
Hi I have a small request, 
I am trying to extend this library to work with a particular radio device.  However, the fixed length format of the protocol I am working with does not include this in the packet and therefore your receive method does not work as intended.

I originally forked your code and changed the method, however, to get better future compatibility I wanted to change to inheritance and building my functionality on top, so that I could benefit from any future improvement you make in the rest of the code.  However, this is not possible at the moment as I cannot create an alternative receive function without access to the SpiReadBurstReg function.

Do you think it may be possible to change this function from private to protected, or alternatively might it be possible to include an alternative recieve function which uses the fixed packet length that has been set instead of assuming it is at the start of the packet.  I am happy to prepare a pull for this if you would be prepared to receive and implement such a new function.  

Regards
Neil

LSatan, 2020-08-26:
hello, yes there is no problem. I will change it. all possible should be exhausted. without restrictions. In any case there will be some updates. to further improve the lib. can you share the link with me? I would like to have a look at the changed reception method. there is always room for improvement.
Regards

coylen, 2020-08-26:
My change was just a quick and dirty hack in the first instance, the packets I am interested in does not have length at the start or status at end, so I introduced a packet length variable which gets set in the packet length function and then this was used in the modified receive, code is attached below, as you will see change was minimal.

`byte ELECHOUSE_CC1101::ReceiveData(byte *rxBuffer)
{
	byte size;
	byte status[2];

	if(SpiReadStatus(CC1101_RXBYTES) & BYTES_IN_RXFIFO)
	{
		//size=SpiReadReg(CC1101_RXFIFO);
		SpiReadBurstReg(CC1101_RXFIFO,rxBuffer,packetLength);
		//SpiReadBurstReg(CC1101_RXFIFO,status,2);
		SpiStrobe(CC1101_SFRX);
    SpiStrobe(CC1101_SRX);
		//return size;
    return pl;
	}
	else
	{
		SpiStrobe(CC1101_SFRX);
    SpiStrobe(CC1101_SRX);
 		return 0;
	}
}`

LSatan, 2020-10-01:
done with v2.5.2