LSatan/SmartRC-CC1101-Driver-Lib #46: the Rc-Switch examples cc1101  works but not the CC1101 default examples

puckpuck85, 2020-08-27:
im using the cc1101 with a esp32 and the example with rcswitch:

"ProtocolAnalyzeDemo_cc1101" works like a charme. i can press on my remote and get the right information in my serial console. also resend the captured code is no problem. power socket goes on and off. 

but these examples does not work: 

cc1101_New_Receive_method_advanced  //  cc1101_Receive_advanced 

same configuration - i could not figure out what i'm doing wrong here.

LSatan, 2020-08-28:
hello, just to avoid misunderstandings. which broadcast example did you use? With the internal receive samples you can only receive the internal send examples from a 2nd cc1101 (cc1101_Transmitt_Hello_World_minimal). if you tried this, there could be a fault in the spi wiring or gdo0 wiring of the transmitter.
For further questions, please do not hesitate to contact me. regards

puckpuck85, 2020-08-29:
hey thanks for your reply :)

this is the example i talked about:

examples/CC1101 default examples/cc1101_New_Receive_method_minimal/cc1101_New_Receive_method_minimal.ino

so this example does need 2 cc1101 modules? one for TX and one for RX?

LSatan, 2020-08-29:
yes the standard examples are used to communicate between 2 cc1101.

puckpuck85, 2020-08-29:
ahhhhh okay. so the best way to monitor the signals arround me with one cc1101 is the :

examples/ESPiLight examples cc1101/Receive_cc1101/Receive_cc1101.ino

ps: thank you sharing your awesome lib with us :)

closed 2026-04-26 by github-actions[bot]