LSatan/SmartRC-CC1101-Driver-Lib #47: Frequency_scanner

DavidAntonin, 2020-09-01:
Hello,

thank you for your work. Can you release new version of Frequency scanner working with new version of library?
I used this utility to find exact transmiting frequency of my remotes and sensors. Than I finetuned receiving frequency of CC1101 together with narrow bandwidth for better range.
Thank you

LSatan, 2020-09-02:
Hi,
Thank you for your interest.
I have re-created the frequency scanner.

``` C++

#include <ELECHOUSE_CC1101_SRC_DRV.h>

float start_freq = 433;
float stop_freq = 434;

float freq = start_freq;
long compare_freq;
float mark_freq;
int rssi;
int mark_rssi=-100;
int compare_scan;
unsigned long time;

void setup() {
Serial.begin(115200);
ELECHOUSE_cc1101.Init();
ELECHOUSE_cc1101.setRxBW(58);
ELECHOUSE_cc1101.SetRx();
Serial.println("Frequency Scanner Ready");
}

void loop() {
ELECHOUSE_cc1101.setMHZ(freq);
rssi = ELECHOUSE_cc1101.getRssi();

if (rssi>-75){
if (rssi > mark_rssi){
mark_rssi = rssi;  
mark_freq = freq;
}
}

freq+=0.01;

if (freq>stop_freq){
freq=start_freq;

if (mark_rssi>-75){

long fr = mark_freq*100;

if (fr == compare_freq){
Serial.print("Freq: ");
Serial.println(mark_freq);
Serial.print("Rssi: ");
Serial.println(mark_rssi);
mark_rssi=-100;
compare_freq = 0;
mark_freq = 0;
}else{
compare_freq = mark_freq*100;
freq = mark_freq -0.10;
mark_freq=0;
mark_rssi=-100;
}
}
}
}

```

Regards

hzhh110, 2020-12-02:
> Hello,
>
> thank you for your work. Can you release new version of Frequency scanner working with new version of library?
> I used this utility to find exact transmiting frequency of my remotes and [...]

Hello, how are you doing? I have been searching for frequency hopping, but the problem of frequency doubling is very serious. We can't distinguish between those [433.92 and 868 < 410 and 820 >...]. Do you have any solutions?

`
float SearchFrequency(float mhz,float freqSpace,boolean isDetails);
float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails);
void setup() {
    delay(3000);
    Serial.begin(115200);
    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setRxBW(58);
    ELECHOUSE_cc1101.SetRx();
    // Serial.println("Frequency Scanner Ready");
    Serial.println("Frequency Scanner Ready");
}

float searchFrequency(float mhz,float freqSpace,boolean isDetails){
    float start_freq = mhz;
    float stop_freq = mhz + 1;
    float freq = mhz;
    long compare_freq;
    float mark_freq;
    int rssi;
    int mark_rssi=-100;
    int searchTime = 0;
    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setMHZ(mhz); 
    ELECHOUSE_cc1101.setRxBW(58.03);
    ELECHOUSE_cc1101.SetRx();
    delay(20);
    while (1)
    {
        searchTime ++;
        if (searchTime > 50)
         {
           return -1;
         }
        ELECHOUSE_cc1101.setMHZ(freq);
        rssi = ELECHOUSE_cc1101.getRssi();  
 
        if (rssi>-75){
            if (rssi > mark_rssi){
                mark_rssi = rssi;  
                mark_freq = freq;
            }
        }
        
        freq+=freqSpace;
        if (freq>stop_freq){
            // Serial.print("freq>stop_freq");
            freq=start_freq;
            if (mark_rssi>-75){
                long fr = mark_freq*100;
                if (fr == compare_freq){
                    Serial.print("freq:");
                    Serial.println(freq);
                    Serial.print("rssi:");
                    Serial.println(rssi);
                    Serial.print("mark_freq: ");
                    Serial.println(mark_freq);
                    Serial.print("searchTime: ");
                    Serial.println(searchTime);
                    float saveMarkFreq  =  mark_freq;
                    mark_rssi=-100;
                    compare_freq = 0;
                    mark_freq = 0;                   
                    if (isDetails)
                    {
                        return saveMarkFreq;
                    }else
                    {   
                        float startSaveMarkFreq = (int)saveMarkFreq-3;
                        float endSaveMarkFreq = (int)saveMarkFreq+3;
                        float  getDetailsRequency =  searchFrequencyForRange(startSaveMarkFreq,endSaveMarkFreq,0.01,true);
                        return getDetailsRequency;
                    }                    
                }else{
                        compare_freq = mark_freq*100;
                        freq = mark_freq -0.10;
                        mark_freq=0;
                        mark_rssi=-100;
                }
         }else{
                return -1;
         }
     }
    }          
}
float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails ){
        while (1)
        {
            float frequency =  searchFrequency(startMhz,freqSpace,isDetails);
            
            if (frequency != -1)
            {
                return frequency;
            }
            startMhz ++;
            if (startMhz >endMhz)
            {
                return -1;
            }
        }
       
}
void loop() {
    float  loopFrequency;
    loopFrequency =  searchFrequencyForRange(300,347,0.1,true);
    Serial.print("loopFrequency:315   ");
    Serial.println(loopFrequency);
    loopFrequency =  searchFrequencyForRange(388,463,0.1,true);
    Serial.print("loopFrequency:433  ");
    Serial.println(loopFrequency);
    loopFrequency =  searchFrequencyForRange(779,927,0.1,true);
    Serial.print("loopFrequency:868   ");
    Serial.println(loopFrequency);
}

`

hzhh110, 2020-12-03:
Hello, boss, I am now encountering <433.92 and 867.8>, <390 and 780> the problem of doubling the frequency. Is there any trick to crack it?

LSatan, 2020-12-06:
Sure... the highest rssi is worth your frequency.
You can exclude the wrong frequencies with rssi threshold.

check rssi_threshold = -35;
you recive nothing? than test -40;
you receive to much? than test -30;
and so on.
if it still does not bring the desired success. Please notify you again.

``` C++

#include <ELECHOUSE_CC1101_SRC_DRV.h>

float start_freq = 433;    //start scan frequency set.
float stop_freq = 434;    //stop scan frequency set.
int rssi_threshold = -75; // set the threshold to react. the value should be between -20 and -90. recommended -75.

float freq = start_freq;
long compare_freq;
float mark_freq;
int rssi;
int mark_rssi=-100;

void setup() {
Serial.begin(115200);
ELECHOUSE_cc1101.Init();
ELECHOUSE_cc1101.setRxBW(58);
ELECHOUSE_cc1101.SetRx();
Serial.println("Frequency Scanner Ready");
}

void loop() {
ELECHOUSE_cc1101.setMHZ(freq);
rssi = ELECHOUSE_cc1101.getRssi();

if (rssi>rssi_threshold){
if (rssi > mark_rssi){
mark_rssi = rssi;  
mark_freq = freq;
}
}

freq+=0.01;

if (freq>stop_freq){
freq=start_freq;

if (mark_rssi>rssi_threshold){

long fr = mark_freq*100;

if (fr == compare_freq){
Serial.print("Freq: ");
Serial.println(mark_freq);
Serial.print("Rssi: ");
Serial.println(mark_rssi);
mark_rssi=-100;
compare_freq = 0;
mark_freq = 0;
}else{
compare_freq = mark_freq*100;
freq = mark_freq -0.10;
mark_freq=0;
mark_rssi=-100;
}
}
}
}

```

danielhfaria, 2023-02-12:
> Sure... the highest rssi is worth your frequency. You can exclude the wrong frequencies with rssi threshold.
>
> check rssi_threshold = -35; you recive nothing? than test -40; you receive to much? than [...]

Hi !! Thank you for your content,  I would like you to help me, in this code it does not work at frequencies lower than 315mhz for example, would you know why?? float start_freq = 314; stop_freq = 315;

danielhfaria, 2023-02-17:
> float SearchFrequency(float mhz,float freqSpace,boolean isDetails); float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails); void setup() { delay(3000); Serial.begin(115200); ELECHOUSE_cc1101.Init(); ELECHOUSE_cc1101.setRxBW(58); ELECHOUSE_cc1101.SetRx(); // Serial.println("Frequency Scanner Ready"); Serial.println("Frequency Scanner Ready"); }
> 
> float searchFrequency(float mhz,float freqSpace,boolean isDetails){ float start_freq = mhz; float stop_freq = mhz + 1; float freq = mhz; long compare_freq; float mark_freq; int rssi; int mark_rssi=-100; int searchTime = 0; ELECHOUSE_cc1101.Init(); ELECHOUSE_cc1101.setMHZ(mhz); ELECHOUSE_cc1101.setRxBW(58.03); ELECHOUSE_cc1101.SetRx(); delay(20); while (1) { searchTime ++; ELECHOUSE_cc1101.setMHZ(freq); rssi = ELECHOUSE_cc1101.getRssi();
> 
> ```
>     if (rssi>-75){
>         if (rssi > mark_rssi){
>             mark_rssi = rssi;  
>             mark_freq = freq;
>         }
>     }
>     
>     freq+=freqSpace;
>     if (freq>stop_freq){
>         // Serial.print("freq>stop_freq");
>         freq=start_freq;
>         if (mark_rssi>-75){
>             long fr = mark_freq*100;
>             if (fr == compare_freq){
>                 Serial.print("freq:");
>                 Serial.println(freq);
>                 Serial.print("rssi:");
>                 Serial.println(rssi);
>                 Serial.print("mark_freq: ");
>                 Serial.println(mark_freq);
>                 Serial.print("searchTime: ");
>                 Serial.println(searchTime);
>                 float saveMarkFreq  =  mark_freq;
>                 mark_rssi=-100;
>                 compare_freq = 0;
>                 mark_freq = 0;                   
>                 if (isDetails)
>                 {
>                     return saveMarkFreq;
>                 }else
>                 {   
>                     float startSaveMarkFreq = (int)saveMarkFreq-3;
>                     float endSaveMarkFreq = (int)saveMarkFreq+3;
>                     float  getDetailsRequency =  searchFrequencyForRange(startSaveMarkFreq,endSaveMarkFreq,0.01,true);
>                     return getDetailsRequency;
>                 }                    
>             }else{
>                     compare_freq = mark_freq*100;
>                     freq = mark_freq -0.10;
>                     mark_freq=0;
>                     mark_rssi=-100;
>             }
>      }else{
>             return -1;
>      }
>  }
> }          
> ```
> 
> } float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails ){ while (1) { float frequency = searchFrequency(startMhz,freqSpace,isDetails);
> 
> ```
>         if (frequency != -1)
>         {
>             return frequency;
>         }
>         startMhz ++;
>         if (startMhz >endMhz)
>         {
>             return -1;
>         }
>     }
> ```
> 
> } void loop() { float loopFrequency; loopFrequency = searchFrequencyForRange(300,347,0.1,true); Serial.print("loopFrequency:315 "); Serial.println(loopFrequency); loopFrequency = searchFrequencyForRange(388,463,0.1,true); Serial.print("loopFrequency:433 "); Serial.println(loopFrequency); loopFrequency = searchFrequencyForRange(779,927,0.1,true); Serial.print("loopFrequency:868 "); Serial.println(loopFrequency); }

Hello buddy, how are you ? did you manage to get this code right? would you sell it? i really need it!!!

closed 2026-04-26 by github-actions[bot]