LSatan/SmartRC-CC1101-Driver-Lib #50: library folder name change

gitterman, 2020-10-23:
I just updated to 2.5.2 and now the library resides under the name of **arduino_392935** instead of the previous folder named **SmartRC-CC1101-Driver-Lib**
Is there a new library naming convention in place or what is the reason for this name change?

LSatan, 2020-10-23:
Hi, that's funny. Nothing was changed. Did you install it through arduino library manager? try to install an older version. then install the new one again. if that doesn't work, delete the folder. restart the ide and reinstall the library. please give feedback i will check it myself later.

Regards

gitterman, 2020-10-23:
Yes I did install using the arduino library manager.
I tried to install an older rev but the library manager still said that the installed version was 2.5.2.
Then I deleted the SmartRC-CC1101-Driver-Lib folder as well as the arduino_392935 folder and installed 2.5.1.
The installation created the folder SmartRC-CC1101...
Then I upgraded to 2.5.2 and it updated the files in that folder.
I think the problem came from an explorer window which I had opened on the SmartRC-CC.. folder when I did the first update.
The update deleted all files in that folder but did not put the new files into it.
Thus, I suppose an update initially downloads the new library to an arbitrary folder named arduino_####### , then deletes the old folder and renames the download folder.
In my case it could only delete the files but not the folder as it was busy and thus, it could not rename the download folder.

closed 2020-10-23 by gitterman