LSatan/SmartRC-CC1101-Driver-Lib #51: possible to use this library with pilight_usb_nano?

orrpan, 2020-10-28:
I've tested a little without success to use this library for https://github.com/pilight/pilight-usb-nano.
Do you think it is possible? To compile I just renamed the `pilight_usb_nano.c` to `pilight_usb_nano.ino`, included your library and removed the function `void delayMicroseconds(unsigned int us)`.
So it compiles but I cannot get it to receive/send or get any sign of it working.

In void setup()
```c

  ELECHOUSE_cc1101.setGDO(5,2);  // ports used by library
  ELECHOUSE_cc1101.Init();
  ELECHOUSE_cc1101.setMHZ(433.92);
  ELECHOUSE_cc1101.SetRx();
```

in void receive() (as I understands it the receiving means in receiving from usb and transmitting, tested both below with tx/rx combinations)
```c
ELECHOUSE_cc1101.SetTx();
```

in void receive()
```c
ELECHOUSE_cc1101.SetRx();
```
Would it be smarter to use the "without_gd0" option?

LSatan, 2020-10-28:
Hi,
the set gdo function is not necessary, as pilight defienes the pins. As I can see the code was written in the atmel studio. Apparently the void loop function is missing. that means it cannot be compiled in Arduino ide.
Which commands do you have to enter in order to send or receive? alternatively you can use the espilight examples from my lib.
Regards

orrpan, 2020-10-28:
Thanks for fast response!
okey, removes it. That seems interesting, since I can compile it in arduino ide.

made a fork. https://github.com/orrpan/pilight-usb-nano/blob/master/pilight_usb_nano_cc1101.ino

I'm not really sure which commands will be sent or received, this works as a bridge between a rf433 antenna and pilight if you not have gipo. But in the code it seems like it something like this:
https://github.com/orrpan/pilight-usb-nano/blob/dcb7cd3b546d83eb72d1642322d593902d6b78f5/pilight_usb_nano_cc1101.ino#L48-L54

LSatan, 2020-10-28:
ok, compiling completed. I will test it later.

orrpan, 2020-12-14:
> ok, compiling completed. I will test it later.

Any luck?

orrpan, 2021-07-05:
https://github.com/latchdevel/pilight-usb-nano seems like @latchdevel got it working in arduinio. I cloned it, added setup and SetRx which make the receiving part work, but I’m not able to Tx. 

I'm guessing I need to work som magic here [tx-part](https://github.com/latchdevel/pilight-usb-nano/blob/0bde571ca706c2cf0a8d8653fe50cde1ea72562e/pilight-usb-nano.ino#L192) . So I tested with setting SetRx and then after SetRx without any result, I also tested using SendData(data) and commenting out all the writeDigital parts. But no success.

Do you have any idea?

(I needed to either send `s:20,200,4000,82000@` or set [this DEFAULT_RX_SETTINGS](https://github.com/latchdevel/pilight-usb-nano/blob/0bde571ca706c2cf0a8d8653fe50cde1ea72562e/pilight-usb-nano.ino#L31) DEFINE to make it work at all)

latchdevel, 2021-07-07:
Hi @orrpan and @LSatan,

RF TX is the simplest part of the whole development, you shouldn't have any trouble getting it to work properly.

The pins of TX and RX must be different. RX PIN should be an [attachable interrupt pin](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/), but TX PIN can be anything that's not in use.

```c++
/ * Configurable RX & TX pins * /
#define RX_PIN 2 // Pin for ASK / OOK pulse input from RF receiver module data output.
#define TX_PIN 5 // Pin for ASK / OOK pulse output to RF transmitter module data input.
````

To transmit any RF code, you just have to write it through the serial port in the correct format, which must have the spaces stripped.
The default configuration tries to emulate the behavior of the original code, but for testing it is recommended to use this:

```c++
//#define EVERY_SEC_LINE_FEED       // If defined, print line feed '\n' every second, to emulate legacy firmware.
#define SEND_STRIPPED_SPACES        // If defined, send every 'space' before 'pulse' in broadcast(), which stripped in legacy firmware.
#define LED_BLINK_RX LED_BUILTIN    // If defined, sets the digital output to blink on valid RF code reception.
#define DEFAULT_RX_SETTINGS         // If defined, sets valid RX settings at boot, like sets 's:20,200,4000,82000@'
#define BOOT_SHOW_SETTINGS          // If defined, show settings at boot, like as: 'v:20,200,4000,82000,2,1,1600@'
#define ADD_LINE_FEED               // If defined, add line feed '\n' each line output.
```

There are a couple of examples [in the code](https://github.com/latchdevel/pilight-usb-nano/blob/0bde571ca706c2cf0a8d8653fe50cde1ea72562e/pilight-usb-nano.ino#L58):

`c: 10202020202020202020202020020202200202200202020202020220020202203;p:279,2511,1395,9486;r:5@`

The PiLight project does not offer much information about the format of the codes and their interpretation, but I have published something about it in this library: **["PiCode Library"](https://github.com/latchdevel/PiCode)** https://github.com/latchdevel/PiCode

And, if you want to be able to generate any code in a simple way, you can use a tool that I have created for it, **["picoder"](https://github.com/latchdevel/picoder)**
 https://github.com/latchdevel/picoder

I will appreciate any comments.
Thanks.