LSatan/SmartRC-CC1101-Driver-Lib #52: What is last argument in SendData()

Scott216, 2020-11-01:
What is the 100 represent in this code?
ELECHOUSE_cc1101.SendData(transmitt_byte, 11, 100);

LSatan, 2020-11-01:
Hi, the 100 stands for a waiting time of 100 milliseconds. Usually the gdo0 pin is used and when sending the gdo0 pin goes high. this means that the code remains in place until the gdo0 pin goes low again. then the code is continued (delete tx fifo) here it is so that we do not use the gdo0 pin. we have to wait a little while before we delete the tx fifo again. to ensure that the code was transmitted. you can also shorten the time by 100 ms. it depends on the data rate and the length of the message. you can try out whether the message was still completely transmitted after a few ms. that brings some speed into your code. 50ms should still work fine. with less easy to test.
Regards

Scott216, 2020-11-01:
Thanks.  I'm only transmitting 10 bytes.   I don't know if my setup even uses the gdo0 pin.  I'm using a panStamp which is like an Arduino Pro Mini with the cc1101 chip attached.  They've been discontinued and their Arduino IDE library isn't playing nice with the latest Arduino IDE.  I just got your example sketches working on my panStamps.

Scott216, 2020-11-02:
[image]

I was looking at the panStamp schematic and I see GDO0 on the cc1101 is connected to PD2(INT0/PCINT18) in the atmega328p.  Would you recommend using your library with GDO?  If so, do I need to configure anything so your library knows what pin to use on atmega328p?

LSatan, 2020-11-02:
you can use the examples with or without gdo (internal send and receive functions only). the reception examples without gdo work better than with. since you can do several things in the loop without missing a message.

this wiring is the default for atmega328:
[image]

the gdo contacts can be changed with this command:
ELECHOUSE_cc1101.setGDO (GDO0, GDO2); 

the spi pins can be changed with this command:
ELECHOUSE_cc1101.setSpiPin (SCK, MISO, MOSI, CSN); 
However, I do not recommend the latter!

personally i would prefer the examples without gdo.