LSatan/SmartRC-CC1101-Driver-Lib #53: How does CC1101 connect with ESP32 and ESP8266

hzhh110, 2020-11-02:
[image]
[image]
How does CC1101 connect with ESP32 and ESP8266? Nothing happens after I connect, do I need to set the setGDO and setSpiPin of CC1101? CC1101 has VCC, GDO0, CSN, SCK, MCSI, GDO1, GDO2

hzhh110, 2020-11-02:
I saw the corresponding definition in the library, the label of that board is wrong, and it can be run.

closed 2020-11-02 by hzhh110

LSatan, 2020-11-02:
Hi, the picture of the esp32 module in the sample drawings shows a different esp32 than yours. you have to pay attention to the wiring here.
[image]

LSatan, 2020-11-02:
you can also use your own pin assignment. the instructions are in readme.md.