LSatan/SmartRC-CC1101-Driver-Lib #55: Put cc1101 in sleep mode

Scott216, 2020-11-02:
My transmitter is running off a battery, so I want to minimize power consumption.  I have a library to put my Arduino to sleep for 8 seconds at a time.  Is there a way to also put the cc1101 chip into sleep mode?

LSatan, 2020-11-02:
the sleep mode comes in the next update. but you can set it manually:
ELECHOUSE_cc1101.SpiStrobe(0x36);//Exit RX / TX, turn off frequency synthesizer and exit
ELECHOUSE_cc1101.SpiStrobe(0x39);//Enter power down mode when CSn goes high.

Scott216, 2020-11-03:
What is CSn?  For the above two lines of code, do I just use one or the other, or both?
How do I come out of sleep mode?
Thanks

LSatan, 2020-11-03:
csn is one of the spi pins. at the moment it should work like this. Note, however, that in the first variant no signals can be received during sleep. with the second variant, however, it should work. but haven't tested it yet. the module also has a wake on radio function. but this will only come with the next update.

``` C++
// enter deep sleep mode
ELECHOUSE_cc1101.SpiStrobe(0x36);//Exit RX / TX, turn off frequency synthesizer and exit

ELECHOUSE_cc1101.SpiStrobe(0x39);//Enter power down mode when CSn goes high.

//wake up
ELECHOUSE_cc1101.SetRx();

```

``` C++
// enter sleep mode

ELECHOUSE_cc1101.SpiStrobe(0x39);//Enter power down mode when CSn goes high.

//wake up
//Nothing

```

Scott216, 2020-11-03:
I'm just putting my transmitter to sleep, it's not receiving anything, so the wake on radio doesn't do anything for me.  I'll try out your first variant.   Thank you.  BTW, I just sent you a donation

Scott216, 2020-11-03:
I did a little current measuring test.  I have a [µCurrent device](https://www.eevblog.com/projects/ucurrent/) from eevblog which comes in handy for measuring small currents
Awake: 6mA
atmega sleep only: 1.7mA
atmega + cc1101 asleep: 0.11 mA

LSatan, 2020-11-03:
Thank you for the donation and the nice comment.☺️

The scores are not bad. Very important in battery operation. 👍

Since you only use the transmission function, SetRx should not be necessary to wake up with the first variant.

Regards

Shekhar-HyperSystems, 2025-08-10:
@LSatan How to implement wake on radio?

closed 2026-04-26 by github-actions[bot]