LSatan/SmartRC-CC1101-Driver-Lib #56: 433.92 and 868

hzhh110, 2020-11-04:
The frequency of 433.92 can be received by the receiving end of 433.92, or it can be received by the receiving end of 868, and the analysis is successful,,for rc-switch

LSatan, 2020-11-04:
sorry i'm not quite sure what you mean.

rc-switch is basically only for am modulation (ask / ook). There are remote controls that send in a different modulation in the 868 MHz range. for example fsk modulation. this cannot be processed before rc-switch.

however, depending on your module, you can send and receive 433mhz version or 868mhz version in all supported frequencies. For 433mhz module 300-348 MHz and 387-464 MHz.
For 868 Module = 779-928 MHz.

I hope this answers your question. if not, please describe again in more detail.

Regards

hzhh110, 2020-11-12:
void copyInitELECHOUSE_cc1101WithMHZ(float mhz){
     ELECHOUSE_cc1101.Init();
     ELECHOUSE_cc1101.setMHZ(mhz);
     ELECHOUSE_cc1101.SetRx(); 
}
void loopStartCopy(){
      while (1){
            for (size_t y = 0; y < numMyMhzDatas; y++)
              {
                copyInitELECHOUSE_cc1101WithMHZ(myMhzDatas[y]);
                mySwitch_R.resetAvailable();
                delay(500);
                if (mySwitch_R.available()){   
                  Serial.print("Received ");
                  Serial.print( mySwitch_R.getReceivedValue() );
                  Serial.print(" / ");
                  Serial.print( mySwitch_R.getReceivedBitlength() );
                  Serial.print("bit ");
                  Serial.print("Protocol: ");
                  Serial.print( mySwitch_R.getReceivedProtocol() );
                  Serial.print("Delay: ");
                  Serial.println( mySwitch_R.getReceivedDelay());
                  Serial.print("mhz: ");
                  Serial.println(myMhzDatas[y]);
                  mySwitch_R.resetAvailable();
                }
              }
      }
}

Received 9985096 / 24bit Protocol: 1Delay: 299
mhz: 433.92

Received 9985096 / 24bit Protocol: 1Delay: 299
mhz: 868.35

I use a remote control with a frequency of 433.92, and sometimes it is 433.92 and sometimes it is 868.35. How do I configure this to stop it?

hzhh110, 2020-11-12:
static const float myMhzDatas[]={433.92,315,868.35};