LSatan/SmartRC-CC1101-Driver-Lib #57: NodeMCU & ProtocolAnalyzeDemo_cc1101

alexliit, 2020-11-11:
Hello
I'm a real noob on the wireless protocols but I'm trying to decode a radio remote on 866 Mhz of a MVC fan system.
After wiring the CC1101 module to the ESP I saw no sings on life on the serial debug.
Then I just pulled out all the CC1101 module ...and I got no error or hangs or other error signs: all the ELECHOUSE_cc1101 comands went well.
Is an expected behaviour or I did something wrong?

Thanks!

LSatan, 2020-11-14:
Hi, sorry for the late reply. protocol analysis demo cannot decode everything. You could record raw data to get something. First of all, it is important to know at which frequency the fan works. brand of fan?
if you don't have the right frequency, it won't receive anything. I will give you raw data example later. first see if you can get the frequency out. is mostly written on the fan or the original remote control. otherwise also in the data sheet of the fan.
Regards

alexliit, 2020-11-15:
Hello LSatan

Yes I'm aware is a bit complex process :-) 

The fan is a "Utek Aura 2" (https://www.utek-air.it/en/controlled-mechanical-ventilation/heat-recovert-units-for-residential-buildings/aura/, the one with the remote).
The frequency is at 866 MHZ: it is not written any were but I found a CC1101 on the fan board and with the help of a SDR I found it.

Thanks!

closed 2026-04-26 by github-actions[bot]