LSatan/SmartRC-CC1101-Driver-Lib #58: Can you introduce a development board that supports multiple frequencies?

hzhh110, 2020-11-14:
Can you introduce a module that supports: 300-348 MHZ, 387-464MHZ /. 779-928MHZ cc1101 module with multiple frequencies?There are restricted modules on the market!

LSatan, 2020-11-14:
Hi, to cover all frequencies. you need to use 2 cc1101. 1 × 433mhz version for 300-348 MHz, 387-464 MHz. 1 × 868mhz version for 779-928Mhz. I can only provide you with a corresponding code and wiring example later. what would you like to receive or send? I mean what examples do you want to use for this?
Regards

hzhh110, 2020-11-15:
Thank you very much！My project is to copy the remote control and then carry out the remote control. I am not sure about the frequency of the other party's remote control, so I need to cover as many frequencies as possible.

hzhh110, 2020-11-21:
> Hi, to cover all frequencies. you need to use 2 cc1101. 1 × 433mhz version for 300-348 MHz, 387-464 MHz. 1 × 868mhz version for 779-928Mhz. I can only provide you with a corresponding code and [...]

Hello, how should 433 and 868 connect with ESP32?

LSatan, 2020-11-22:
hi, 
sorry for the late feedback.
you have to do it like here:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/21#issuecomment-573484045

closed 2026-04-26 by github-actions[bot]