LSatan/SmartRC-CC1101-Driver-Lib #59: Reading 345 mhz from honeywell security system

vsajit, 2020-11-17:
I have been trying to read 345 mhz signal from honeywell home security system using cc1101; all I could read is the Rssi and LQI values using the example file cc1101_Receive_minimal. I tried to receive 433.92 Mhz as well, same result. 
ELECHOUSE_cc1101.ReceiveData(buffer); does not return anything

Any suggestions? The goal to read 345 Mhz signals.. I am able to read 433 Mhz completely (including payload) using the example with RC Switch from the cc1101 module, hence the hardware connections seems all right
Thanks in advance for help

LSatan, 2020-11-22:
hi,
sorry for the late feedback.
the reason that it doesn't work is simply that he doesn't know the protocol.
they should try a raw data scanner.
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/44#issuecomment-693636925

vsajit, 2020-11-23:
Thanks a lot for the response.. I will check and post the results

sandstormkeshav, 2023-01-12:
> Thanks a lot for the response.. I will check and post the results

@vsajit did you have any luck with this?