LSatan/SmartRC-CC1101-Driver-Lib #60: I can't get work, help

rferrari81, 2020-11-19:
hi everyone, i can't get CC1101 to work with EP32.
I am attaching the pins I used on the ESP32,

BUZZER		PIN 13

SDCARD	SD_MOSI	PIN 23
	SD_SCK	PIN 18
	SD_MISO	PIN 19
	SD_CS	PIN 12

LCD	VCC	VCC
	GND	GND
	CS	PIN 17
	RESET	PIN 5
	D/C	PIN 16
	MOSI	PIN 23
	SCK	PIN 18
	LED	PIN 32
	MISO	PIN 19
	T_CLK	PIN 18
	T_CS	PIN 21
	T_DI	PIN 23
	T_DO	PIN 19

IR	RX	PIN 14
	TX	PIN 15

CC1101	VCC	VCC
	GND	GND
	CSN	27
	MOSI	23
	MISO	19
	SCK	18
	GD0	26
	GD2	25

I can't start the sketch with dislpay and CC1101, it gives me double SPI error. starting it only with CC1101 nothing works, even the serial monitor does not start.
Please help me!!!

LSatan, 2020-11-22:
hi,
sorry for the late feedback. use a different spi interface for cc1101. 
use the following command and wiring in setup before init:

ELECHOUSE_cc1101.setSpiPin(14,12,13,15); //SCK, MISO, MOSI, CSN

notpike, 2021-02-13:
Hi!

I'm having same issue with sharing the CC1101 with other devices on the same SPI bus. 

I'm trying to make this lib work with the M5Stack which is an ESP32 that uses a ILI9341 display and SD card. Both the display and SD card uses the same SPI bus. I tried making the library to make this coexist with the other two devices with no luck .

Would you have any advice on how to make this work?

Thanks!
-NotPike

BG7UHS, 2021-09-20:
I also encountered the same problem. 
[image]