LSatan/SmartRC-CC1101-Driver-Lib #61: 868mhz and 433.92mhz

hzhh110, 2020-11-26:
When receiving 868MHZ, how to block 433.92MHZ and how to set it? After the remote controllers of 433.92 and 868MHZ are sent, 868MHZ is received normally, and it is impossible to determine which frequency it is.

hzhh110, 2020-11-26:
I now use a CC1101 module, which can transmit 315, 433, 868, 915 frequency bands, but I am not sure whether the remote control I copied is 433.92MHz or 868MHz

hzhh110, 2020-11-26:
I tried ELECHOUSE_cc1101.setRxBW(200); and found that the 433.92 frequency can be blocked. What is the principle?????

LSatan, 2020-11-26:
Hi,
why do you want to block 433.92? you receive what you set. if you only want to receive 868 mhz then set it and only 868 mhz will be received. if you receive something at 868mhz and 433.92 mhz then this is not necessarily unusual. if you are too close to the receiver, for example, another frequency can break through. I suspect, however, that your remote control sends 868 and 433 on both frequencies. This serves as a backup and is often used in alarm systems. to ensure that the signal arrives at the receiver, it is sent dual. rxbw is the reception bandwidth! that means if you set 433.92 mhz and rxbw = 200hz. then you will only receive between 433.82 and 434.02

Regards

hzhh110, 2020-11-27:
Thank you very much, because I am here for the copy of the remote control, and each remote has a corresponding frequency, and we don't know the corresponding frequency, so first I must confirm the corresponding frequency here!Also, is there some way to quickly locate the frequency? For the time being, I can only achieve a fixed copy of a few frequencies. If I can determine the frequency first, I can copy according to the frequency.

hzhh110, 2020-11-27:
How to determine the large range first<300-348 MHZ, 387-464MHZ and 779-928MHZ>, and then determine the small range? How to quickly get the corresponding frequency?

hzhh110, 2020-11-27:
When I keep pressing the remote control to transmit data of unknown frequency, how can I quickly locate the frequency of the remote control?

LSatan, 2020-11-27:
have you ever tried the frequency scanner. you need to set the scan area. I do not recommend scanning more than 1mhz. you can also scan 2 MHz but it takes longer.
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/47#issuecomment-685908390

hzhh110, 2020-11-28:
`#include <Arduino.h>
#include "RCSwitch.h"
#include "ELECHOUSE_CC1101_SRC_DRV.h"

float SearchFrequency(float mhz,float freqSpace,boolean isDetails);
float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails);
void setup() {
    delay(3000);
    Serial.begin(115200);
    ELECHOUSE_cc1101.Init();
    // ELECHOUSE_cc1101.setMHZ(315);
    ELECHOUSE_cc1101.setRxBW(58);
    ELECHOUSE_cc1101.SetRx();
    Serial.println("Frequency Scanner Ready");
}

float SearchFrequency(float mhz,float freqSpace,boolean isDetails){

  float start_freq = mhz;
  float stop_freq = mhz + 1;
  float freq = mhz;
  long compare_freq;
  float mark_freq;
  int rssi;
  int mark_rssi=-100;
    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setMHZ(mhz);
    ELECHOUSE_cc1101.setRxBW(58.03);
    ELECHOUSE_cc1101.SetRx();
 //312  and  878-879  have problem
    if (mhz == 311)
    {
        return -1;
    }
    if (mhz == 878)
    {
       return -1;
    }
    while (1)
    {
        ELECHOUSE_cc1101.setMHZ(freq);
        rssi = ELECHOUSE_cc1101.getRssi();

       
        if (rssi>-75){
            if (rssi > mark_rssi){
                mark_rssi = rssi;  
                mark_freq = freq;
            }
        }
        freq+=freqSpace;
        if (freq>stop_freq){
            freq=start_freq;
            if (mark_rssi>-75){
                long fr = mark_freq*100;
                if (fr == compare_freq){
                    Serial.print(freq);
                    Serial.print(":");
                    Serial.print(rssi);
                    Serial.print(":");
                    Serial.println(ELECHOUSE_cc1101.getLqi());

                    Serial.print("Freq: ");
                    Serial.println(mark_freq);
                    float saveMarkFreq  =  mark_freq;
                    mark_rssi=-100;
                    compare_freq = 0;
                    mark_freq = 0;
                   
                    if (isDetails)
                    {
                        return saveMarkFreq;
                    }else
                    {   
                        float startSaveMarkFreq = (int)saveMarkFreq-3;
                        float endSaveMarkFreq = (int)saveMarkFreq+3;
                        float  getDetailsRequency =  searchFrequencyForRange(startSaveMarkFreq,endSaveMarkFreq,0.01,true);
                        return getDetailsRequency;
                    }                    
                }else{
                        compare_freq = mark_freq*100;
                        freq = mark_freq -0.10;
                        mark_freq=0;
                        mark_rssi=-100;
                }
         }else{
                return -1;
         }
     }
    }          
}
float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails ){
        while (1)
        {
            float frequency =  SearchFrequency(startMhz,freqSpace,isDetails);
            
            if (frequency != -1)
            {
                return frequency;
            }
            startMhz ++;
            if (startMhz >endMhz)
            {
                return -1;
            }
        }
       
}
//can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet
void loop() {
    float  loopFrequency;
    loopFrequency =  searchFrequencyForRange(387,463,0.1,true);
    Serial.print("loopFrequency:433   ");
    Serial.println(loopFrequency);

    loopFrequency =  searchFrequencyForRange(779,927,0.1,true);
    Serial.print("loopFrequency:868   ");
    Serial.println(loopFrequency);
}
`

hzhh110, 2020-11-28:
Hello, thank you very much for your help, can you help me optimize it? I have a few problems with the code here
1, I get the corresponding frequency by traversing the loop, but it repeats every time
ELECHOUSE_cc1101.Init();
ELECHOUSE_cc1101.setMHZ(mhz);
ELECHOUSE_cc1101.setRxBW(58.03);
ELECHOUSE_cc1101.SetRx();
Otherwise, it won't be accurate, how can this process be optimized?
void ELECHOUSE_CC1101::Init(void)
{
   setSpi();
   SpiStart(); //spi initialization
   digitalWrite(SS_PIN, HIGH);
   digitalWrite(SCK_PIN, HIGH);
   digitalWrite(MOSI_PIN, LOW);
   Reset(); //CC1101 reset
   RegConfigSettings(); //CC1101 register config
   SpiEnd();
}

2. Every time you send 390---464, the frequency multiplier will also respond <779-928>. For example, when 433.92 and 868 will respond, how to deal with this problem?

hzhh110, 2020-11-28:
#include <Arduino.h>
#include "RCSwitch.h"
#include "ELECHOUSE_CC1101_SRC_DRV.h"

float SearchFrequency(float mhz,float freqSpace,boolean isDetails);
float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails);
void setup() {
    delay(3000);
    Serial.begin(115200);
    ELECHOUSE_cc1101.Init();
    // ELECHOUSE_cc1101.setMHZ(315);
    ELECHOUSE_cc1101.setRxBW(58);
    ELECHOUSE_cc1101.SetRx();
    Serial.println("Frequency Scanner Ready");
}

float SearchFrequency(float mhz,float freqSpace,boolean isDetails){

  float start_freq = mhz;
  float stop_freq = mhz + 1;
  float freq = mhz;
  long compare_freq;
  float mark_freq;
  int rssi;
  int mark_rssi=-100;
    ELECHOUSE_cc1101.Init();
    ELECHOUSE_cc1101.setMHZ(mhz);
    ELECHOUSE_cc1101.setRxBW(58.03);
    ELECHOUSE_cc1101.SetRx();

    //312  and  878-879  have problem
    if (mhz == 311)
    {
        return -1;
    }
    if (mhz == 878)
    {
       return -1;
    }
    while (1)
    {
        ELECHOUSE_cc1101.setMHZ(freq);
        rssi = ELECHOUSE_cc1101.getRssi();

       
        if (rssi>-75){
            if (rssi > mark_rssi){
                mark_rssi = rssi;  
                mark_freq = freq;
            }
        }
        freq+=freqSpace;
        if (freq>stop_freq){
            freq=start_freq;
            if (mark_rssi>-75){
                long fr = mark_freq*100;
                if (fr == compare_freq){
                    Serial.print(freq);
                    Serial.print(":");
                    Serial.print(rssi);
                    Serial.print(":");
                    Serial.println(ELECHOUSE_cc1101.getLqi());

                    Serial.print("Freq: ");
                    Serial.println(mark_freq);
                    float saveMarkFreq  =  mark_freq;
                    mark_rssi=-100;
                    compare_freq = 0;
                    mark_freq = 0;
                   
                    if (isDetails)
                    {
                        return saveMarkFreq;
                    }else
                    {   
                        float startSaveMarkFreq = (int)saveMarkFreq-3;
                        float endSaveMarkFreq = (int)saveMarkFreq+3;
                        float  getDetailsRequency =  searchFrequencyForRange(startSaveMarkFreq,endSaveMarkFreq,0.01,true);
                        return getDetailsRequency;
                    }                    
                }else{
                        compare_freq = mark_freq*100;
                        freq = mark_freq -0.10;
                        mark_freq=0;
                        mark_rssi=-100;
                }
         }else{
                return -1;
         }
     }
    }          
}
float searchFrequencyForRange(float startMhz,float endMhz,float freqSpace,boolean isDetails ){
        while (1)
        {
            float frequency =  SearchFrequency(startMhz,freqSpace,isDetails);
            
            if (frequency != -1)
            {
                return frequency;
            }
            startMhz ++;
            if (startMhz >endMhz)
            {
                return -1;
            }
        }
       
}
//can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet
void loop() {
    float  loopFrequency;
    loopFrequency =  searchFrequencyForRange(387,463,0.1,true);
    Serial.print("loopFrequency:433   ");
    Serial.println(loopFrequency);

    loopFrequency =  searchFrequencyForRange(779,927,0.1,true);
    Serial.print("loopFrequency:868   ");
    Serial.println(loopFrequency);
}