LSatan/SmartRC-CC1101-Driver-Lib #64: "Repeater_cc1101.ino" only received and transmitted the first time then afterwards it has been unresponsive.

arthanth, 2021-01-08:
Has anybody worked with "Repeater_cc1101.ino" ?
I tested "ReceiveDemo_Simple_cc1101.ino" and "SendDemo_cc1101.ino" on Arduino Nano ATmega328P it worked fine; however, the Repeater_cc1101.ino" has problem. It only worked the first time. The first time it both received and transmitted afterwards it has been unresponsive.

LSatan, 2021-01-08:
hi, i will change and improve the sketch again. it was one of my first additional skits for rc-switch. now it can be simplified and made shorter. regards

arthanth, 2021-01-09:
Hello LSatan,
Thank you  very much for fast reply.
I tested the "Repeater_cc1101.ino" on NodeMCU-32S ESP32 it worked fine; however ; "Repeater_cc1101.ino" does not work on Arduino Nano ATmega328P.
(tested by SmartRC-CC1101-Driver-Lib_V2.5.2)
Thanks.

LSatan, 2021-01-09:
Please check this and give me feedback.

``` C++

/*
  Simple example for repeating decimal code for cc1101
  
  https://github.com/sui77/rc-switch/
  https://github.com/LSatan/SmartRC-CC1101-Driver-Lib
  ----------------------------------------------------------
  Mod by Little Satan. Have Fun!
  ----------------------------------------------------------
*/
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>

int pinRx; // int for Receive pin.
int pinTx; // int for Transmit pin.

RCSwitch mySwitch = RCSwitch();

int number = 15;  // set number of transmission repetitions.
int TIME = 3000; // set delay befor repeat. For direct repetition after receive set 0.

void setup() {
  Serial.begin(9600);

#ifdef ESP32
pinRx = 4; pinTx = 2;  // for esp32! Receiver on GPIO pin 4. Transmit on GPIO pin 2.
#elif ESP8266
pinRx = 4; pinTx = 5;  // for esp8266! Receiver on pin 4 = D2. Transmit on pin 5 = D1.
#else
pinRx = 0; pinTx = 6;  // for Arduino! Receiver on interrupt 0 => that is pin #2. Transmit on pin 6.
#endif   

//CC1101 Settings:                (Settings with "//" are optional!)
  ELECHOUSE_cc1101.Init();            // must be set to initialize the cc1101!
//ELECHOUSE_cc1101.setRxBW(812.50);  // Set the Receive Bandwidth in kHz. Value from 58.03 to 812.50. Default is 812.50 kHz.
//ELECHOUSE_cc1101.setPA(10);       // set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12)   Default is max!
  ELECHOUSE_cc1101.setMHZ(433.92); // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
  
  ELECHOUSE_cc1101.SetRx();       // set Receive on
  mySwitch.enableReceive(pinRx); // Receiver on

}
void loop() {

  if (mySwitch.available()){
    
    Serial.print("Received ");
    Serial.print( mySwitch.getReceivedValue() );
    Serial.print(" / ");
    Serial.print( mySwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.print( mySwitch.getReceivedProtocol() );
    Serial.print(" Delay: ");    
    Serial.println( mySwitch.getReceivedDelay() );

    delay(TIME);   
    mySwitch.disableReceive();                                                        // Receiver off
    mySwitch.enableTransmit(pinTx);                                                   // Transmit on
    ELECHOUSE_cc1101.SetTx();                                                         // set Transmit on

    Serial.println("Transmit");

    mySwitch.setRepeatTransmit(number);                                               // transmission repetitions.
    mySwitch.setProtocol(mySwitch.getReceivedProtocol());                             // send Received Protocol
    mySwitch.setPulseLength(mySwitch.getReceivedDelay());                             // send Received Delay
    mySwitch.send(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength());      // send Received value/bits
  
    ELECHOUSE_cc1101.SetRx();                                                         // set Receive on
    mySwitch.disableTransmit();                                                       // set Transmit off
    mySwitch.enableReceive(pinRx);                                                    // Receiver on

    Serial.println("Receive");
    
    mySwitch.resetAvailable();

   }
}

```

arthanth, 2021-01-09:
The result shown

Received 13964736 / 24bit Protocol: 1 Delay: 393
Transmit
Receive

only first time afterwards it has been unresponsive.

LSatan, 2021-01-12:
hey sorry saw the answer too late I'll take care of it. which cc1101 module do you use?

arthanth, 2021-01-18:
https://www.analogread.com/product/200/cc1101-wireless-module

I use the CC1101 module. I connected the CC1101 module with ATmega328P and NodeMCU-32S ESP32 as your guide in SmartRC-CC1101-Driver-Lib-master\img folder.

LSatan, 2021-01-19:
hey I tested the old and the new variant on the nano again! It works without any problems! I use the same cc1101 module. I don't know what you're doing wrong. please check if you have swapped GDO0 and GDO2. this is the most common error.

arthanth, 2021-01-25:
Thank you very much LSatan. I am ordering a new cc1101 module.
I will let you know the results when I have tested the new one.

arthanth, 2021-02-09:
https://www.aliexpress.com/item/32962150714.html?srcSns=jp.naver.line.android&spreadType=socialShare&bizType=ProductDetail&social_params=20061990554&tt=MG&aff_fsk=_mPDT6S3&aff_platform=default&sk=_mPDT6S3&aff_trace_key=21660e6df0fb474692d772a029d261c2-1612828339827-07463-_mPDT6S3&shareId=20061990554&businessType=ProductDetail&platform=AE&terminal_id=333f952f1ff9499db0c151c5d24fdde6

https://www.aliexpress.com/item/33059343309.html?invitationCode=UjdFSWxCS2VZeERZbzJVSFpFcHI5eUtGUVBka2Noa2ZWTSsxTDJLUitJWExERU1aVXozNm9RPT0&srcSns=jp.naver.line.android&spreadType=productdetail&bizType=sellerinvite&social_params=20062617473&spreadCode=UjdFSWxCS2VZeERZbzJVSFpFcHI5eUtGUVBka2Noa2ZWTSsxTDJLUitJWExERU1aVXozNm9RPT0&tt=MG&aff_fsk=_mPILTC7&aff_platform=default&sk=_mPILTC7&aff_trace_key=b6893ee3873a4632a6f1b5faa5bd45be-1612828340754-08866-_mPILTC7&shareId=20062617473&businessType=ProductDetail&platform=AE&terminal_id=333f952f1ff9499db0c151c5d24fdde6

I got new 2 modules. (the 2 web links above) I tested them both with the "Repeater_cc1101.ino" on Arduino Nano ATmega328P. They both worked.

FlipB, 2021-07-29:
I ran into this very same issue with the repeater sketch (Receive and transmit works, but after first transmit no new messages can be received).
I am using [this module from aliexpress](https://www.aliexpress.com/item/4001297422884.html?spm=a2g0o.store_pc_groupList.8148356.34.7bd26b7eEGDpDx).

I seem to get this to work reliably by toggling Rx and Tx mode a couple of additional times, eg instead of:

```c
mySwitch.send(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength());
  
ELECHOUSE_cc1101.SetRx();
mySwitch.disableTransmit();
mySwitch.enableReceive(pinRx);

Serial.println("Receive");
mySwitch.resetAvailable();
```

I do
```c
mySwitch.send(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength());

mySwitch.disableTransmit();
ELECHOUSE_cc1101.SetRx();
ELECHOUSE_cc1101.SetTx();
ELECHOUSE_cc1101.SetRx();
mySwitch.enableReceive(rxPin);

Serial.println("Receive");
mySwitch.resetAvailable();
```

arroyorl, 2024-09-25:
> I ran into this very same issue with the repeater sketch (Receive and transmit works, but after first transmit no new messages can be received). I am using [this module from [...]

I have same issue with an ESP8266, and using this FlipB workarround, it works well. Thanks