LSatan/SmartRC-CC1101-Driver-Lib #67: Use of the i2c on wemos d1 mini

boumboum68, 2021-01-27:
Hi there

I'm struggling to find a solution for my question perhaps you could help me... 😅
The gdo0 and 2 pins are connected to the i2c pin on the wemos
I wold like to connect a oled display to display info
How could I connect this 2 pins and configure them to be able to use the i2c?.
Thanks for your help

LSatan, 2021-01-27:
hi, you still have D0, D3 and D4 to choose from for the gdo pins. change the pins in the sketch and try whether sending and receiving still works. Regards

boumboum68, 2021-01-27:
Thanks @LSatan  I've tried but impossible to program the wemos module
Does any other configuration of the pin have to be done or only the d pin on the cc1101 library?
Tha KS

LSatan, 2021-01-27:
no you have to change the pin in the examples. which example do you use?

boumboum68, 2021-01-27:
OK
This is the program I use or trying 😅
https://github.com/supermomo31/Somfy_MQTT
But can't find where to change only in your library

boumboum68, 2021-01-27:
Perhaps you better find the declaration that could save me

LSatan, 2021-01-27:
You do not understand. it is changed in the skitze. not in my lib. in this case the pin is changed in somfy_remote.cpp. 

[image]

regards

boumboum68, 2021-01-27:
yes exactelly what i've done to have the DO and D3 to be used 
[image]
but then the module have no output on the serial monitor like it hangs .
[image]

boumboum68, 2021-01-27:
if i remove one of the GDO pins then it works normally that's why my question:
[image]

LSatan, 2021-01-27:
Ok and 2 and 16?

LSatan, 2021-01-27:
0 is use for flash

boumboum68, 2021-01-27:
right i chaznge it to D4 meaning 2 and D0 16
[image]
perhaps do iu do something wrong on the pining numer?

LSatan, 2021-01-27:
16 was Used for wake up

LSatan, 2021-01-27:
Is it only transmit?

boumboum68, 2021-01-27:
yes only transmit

LSatan, 2021-01-27:
ok i think it can only send. I will later make a few corrections to the somfy_remote.cpp. then gdo0 on D4. you won't need gdo2.

boumboum68, 2021-01-27:
oh thanks thats really cool ..
then waiting for your feedback .

LSatan, 2021-01-27:
I wonder a little. gdo2 can actually only receive and not send. at least in ask mode. gdo0 can do both.

boumboum68, 2021-01-27:
So this mean that gdo0 have to used? Or not possible at all?

LSatan, 2021-01-27:
> So this mean that gdo0 have to used? Or not possible at all?

True we use gdo0.

> Do you think that better is to use a esp32 module in place of the wemos d1 mini?

Esp32 is better. Now we check wemos d1 At First

boumboum68, 2021-01-27:
> Do you think that better is to use a esp32 module in place of the wemos d1 mini? 

OK boss 👌😅

LSatan, 2021-01-27:
ok, check the somfy_mqtt.zip. Set the TxPin in the somfy_mqtt.ino after all the includes. give me a feedback on whether it works.
[Somfy_MQTT.zip](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/files/5882921/Somfy_MQTT.zip)

boumboum68, 2021-01-27:
@LSatan thanks a lot.. That will cost me a beer 😉
I wil' try this tommorow morning and be back to you.
By the way seems that even with the standard config when you remove the gdo2 wire seems working in TX only but not sure that the right way.

boumboum68, 2021-01-29:
I'm trying to test you code but i have an error in the compile:
[image]

LSatan, 2021-01-29:
sorry I can only check it on Sunday. since i'm not at home. I'll sign up again.

boumboum68, 2021-01-29:
Absolutely no problem.. Have a nice weekend