LSatan/SmartRC-CC1101-Driver-Lib #69: CC1101 with Nodemcu - short Rx range with Blynk

miost1, 2021-02-18:
Hello

I'm writng a code for small security system, with some commercially available PIR 433 Mhz sensors (solid code) and CC1101 with ESP8266 Nodemcu as a central reciever. I use the Blynk as a platform. In Blynk it is recommended to use timer function to create loop that controls incoming 433 signals.

I found unusual behavior of the system, when I try to integrate several PIR sensors. The Rx range is very short, about 2 meters. When I use Arduino Nano with simple default code without Blynk I can integrate PIR sensors without any problem with Rx range.

Is Blynk generate the problem with communication range ?

LSatan, 2021-02-18:
hi, I am familiar with blynk and have already used it. I could imagine the blynk routine causing delays. but normally the interrupt routine for the reception runs in parallel. which library do you use to receive the sensors? regards

miost1, 2021-02-18:
The library is yours SmartRC-CC1101-Driver-Lib. I wonder if is it a Blynk problem or maybe somekind of interferation between Nodemcu, power Supply and CC1101. Nevertheless if there is no Wifi in Arduino Nano everything is OK.

LSatan, 2021-02-18:
I actually meant a library like rc-switch.

miost1, 2021-02-18:
Of course the Rc-switch is also used.

LSatan, 2021-02-18:
can you show me your sketch?

miost1, 2021-02-18:
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <RCSwitch.h>
#include <ELECHOUSE_CC1101_SRC_DRV.h>

RCSwitch mySwitch = RCSwitch();
BlynkTimer timer;
char auth[] = "xxxx";

int pin; // int for Receive pin.

void rf433 () {
  if (mySwitch.available()) {

        if (mySwitch.getReceivedValue() == 13270148) {
      //delay (100);
      Blynk.virtualWrite(V1, 1);
      Blynk.virtualWrite(V2, (ELECHOUSE_cc1101.getRssi()));
    }
    else if (mySwitch.getReceivedValue() == 13650702) {
      //delay (100);
      Blynk.virtualWrite(V3, 1);
      Blynk.virtualWrite(V4, (ELECHOUSE_cc1101.getRssi()));
    }
    
delay (100);

    Serial.print("Received ");
    Serial.print( mySwitch.getReceivedValue() );
    Serial.print(" / ");
    Serial.print( mySwitch.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( mySwitch.getReceivedProtocol() );
    Serial.print("RSSI: ");
    Serial.println(ELECHOUSE_cc1101.getRssi());

    mySwitch.resetAvailable();

  }
  else

  { Blynk.virtualWrite(V1, 0);
    Blynk.virtualWrite(V3, 0);
   delay (100);
  }}
  void setup() {
    Blynk.begin(auth, "xxx", "xxxx");
    timer.setInterval(2000L, rf433 );
    Serial.begin(9600);

#ifdef ESP32
    pin = 4;  // for esp32! Receiver on GPIO pin 4.
#elif ESP8266
    pin = 4;  // for esp8266! Receiver on pin 4 = D2.
#else
    pin = 0;  // for Arduino! Receiver on interrupt 0 => that is pin #2
#endif

     if (ELECHOUSE_cc1101.getCC1101()){       // Check the CC1101 Spi connection.
      Serial.println("Connection OK");
      }else{
      Serial.println("Connection Error");
      }

    //CC1101 Settings:                (Settings with "//" are optional!)
    ELECHOUSE_cc1101.Init();            // must be set to initialize the cc1101!
    //ELECHOUSE_cc1101.setRxBW(812.50);  // Set the Receive Bandwidth in kHz. Value from 58.03 to 812.50. Default is 812.50 kHz.
    //ELECHOUSE_cc1101.setPA(10);       // set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12)   Default is max!
    ELECHOUSE_cc1101.setMHZ(433.92); // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.

    mySwitch.enableReceive(pin);  // Receiver on

    ELECHOUSE_cc1101.SetRx();  // set Receive on

  }
  void loop() {

    Blynk.run();
    timer.run();
  }

miost1, 2021-02-18:
What is strange, that serial printing improves the situation. When I remove all the serial printings it is getting worse