LSatan/SmartRC-CC1101-Driver-Lib #72: Calibration problem

kaczorgsm, 2021-03-22:
Hi.
I have problem with enable TX.
Calibration frequency example.
ESP32.

look into code:

**_ELECHOUSE_cc1101.setGDO(gdo0,0);_**   --->

void ELECHOUSE_CC1101::setGDO(byte gdo0, byte gdo2){
GDO0 = gdo0;
GDO2 = gdo2;  
GDO_Set();
}   ---->

void ELECHOUSE_CC1101::GDO_Set (void)
{
	**pinMode(GDO0, INPUT);**
	pinMode(GDO2, OUTPUT);
}

So we have GDO0 as INPUT but for transmit You use:

void transmitt (){
ELECHOUSE_cc1101.SetTx();
for (int i = 0; i<100; i++){
**digitalWrite(gdo0,HIGH);**
delayMicroseconds(500);
**digitalWrite(gdo0,LOW);**
delayMicroseconds(100);
}
ELECHOUSE_cc1101.SetRx();
}

So IMHO this is problem :(
Also please write more details about IOCFG0 registry. Value 0x06 is OK ?
Or maybe we must switch to async mode and set 0x0D ?

LSatan, 2021-03-23:
hello, the set gdo function is intended for the internal transmission functions.
try the example
``` C++

#include <ELECHOUSE_CC1101_SRC_DRV.h>

int gdo0 = 2; // int for Transmit pin.

void setup() {
Serial.begin(9600);
pinMode(gdo0,OUTPUT);
ELECHOUSE_cc1101.Init(); 
ELECHOUSE_cc1101.setMHZ(433.92);
}

void loop(){

ELECHOUSE_cc1101.SetTx();
for (int i = 0; i<100; i++){
digitalWrite(gdo0,HIGH);
delayMicroseconds(500);
digitalWrite(gdo0,LOW);
delayMicroseconds(100);
}
ELECHOUSE_cc1101.SetRx();
}
```

Regards

kaczorgsm, 2021-03-23:
Yep.
This is example from link :)
They have bug :)

But problem is solved. Tnx

PS. Sorry. I wrote in wrong place. correct be https://github.com/LSatan/CC1101-Debug-Service-Tool

closed 2021-03-23 by kaczorgsm