LSatan/SmartRC-CC1101-Driver-Lib #78: Recommendation - On ESP32 don't connect gpio2 to gd0

NorthernMan54, 2021-04-20:
I'm working thru getting fsk demodulation working in asynchronous mode for the rtl_433 port to the esp32 ( https://github.com/NorthernMan54/rtl_433_ESP/issues/5 ) and was looking at leveraging carrier sense feature to determine start and end of signal for processing.

I went and enabled gd0 to be in carrier sense mode

```
SpiWriteReg(CC1101_IOCFG0,      0x0E);   // Carrier sense
SpiWriteReg(CC1101_AGCCTRL1, 0x10);   // Carrier sense relative +6db
```

And found that after flashing this configuration to my device I could no longer **no longer flash** it again.  After much digging into why I couldn't flash the device I found this

https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#gpio2

Based on my read of that, I believe what is happening is that the cc1101 is holding gpio2 high, and not allowing it to enter the serial bootloader.  I then went and disconnected gpio2 and was able to successfully flash my esp32 again.

Also on a esp32 dev board, ( atleast on the ones I'm using ), gpio2 is also the LED

Looking at my layout, I'm thinking we should be recommending something like gpio25 instead. Looking at the available pins on the right side of a devboard,  gpio15 is in pwm mode at boot and gpio21 and gpio22 are used by the i2c connection so am thinking will need to use a gpio from the left side.

[image]

NorthernMan54, 2021-04-26:
Spent yesterday with the soldering iron reworking my units and ended up using gpio13 instead of gpio25 due to wire lengths in my units. Revised diagram looks like this

[image]

I running OpenMQTTGateway on this, and have it also sharing data from a BME280 Temperature Sensor and a BH1750 Light Sensor.

And a reworked unit looks like this ( This one doesn’t have the BH1750, but I left an opening to connect it beside the bme280

[image]

LSatan, 2021-04-26:
hi, great project. Personally, I haven't seen any problems with gpio2 yet. As for the upload. The wiring examples also only show the preset standard of the lib. Of course, everyone is free to change the pins. That's why there are setSpi and set gdo or set rx / tx functions to change the pins as you wish.
Regards

NorthernMan54, 2021-04-26:
Okay, will close this and leave it as reference for others.

closed 2021-04-26 by NorthernMan54