LSatan/SmartRC-CC1101-Driver-Lib #80: RC-Switch SendDemo_cc1101 not working for me

Raynman77, 2021-07-21:
I wired the CC1101 module to an Uno with the same diagram as the Nano pinout and uploaded the RC-Switch cc1101 example "SendDemo_cc1101". The only modifications I made to the example were changing the frequency to `ELECHOUSE_cc1101.setMHZ(315.00);` and changed the binary value to send to `mySwitch.send("000000000001010100010100");` and then uploaded the sketch to the Uno. It doesn't work. The serial monitor reports the connection is "OK". I've used the RC-Switch library alone in the past with a 315mhz transmitter sending this same binary value and it worked just fine.

LSatan, 2021-07-21:
if it works with a 315 mhz transmitter then it should also work with cc1101. the most common error is that gdo0 and gdo2 have been swapped. please check and give me feedback. if you did it with a breadboard, there may also be faulty connections. even if spi ok is displayed. I will improve the spi output later so that it can be ruled out 100%. at the moment it is only 80%. but please check first whether gdo have not been exchanged.

Raynman77, 2021-07-21:
gdo2 receives on pin 2 of the arduino correct? I have gdo2 of the cc1101 module attached to pin 2 of the Uno and when I upload the RC-Switch example "repeater_cc1101", then open a serial monitor and press the button on a known 315mhz remote control, I get the correct protocol and value printed onto the monitor. That makes me believe that gdo0 and gdo2 are in the correct position.

LSatan, 2021-07-21:
yes gdo2 on pin2 is correct for rx. but if you can receive you must not assume that the gdo pins are correct. since gdo0 can do both rx / tx.

Raynman77, 2021-07-21:
Understood. I swapped gdo0 and gdo2 and reset the Uno. Still won't work. I'm not using a breadboard in case you were wondering. I have jumpers that are female to male dupont connectors and I have ohm'd them out and they all passed.

Raynman77, 2021-07-22:
After I decided to leave out the pulse duration and signal repeat, the receiving end of the 315mhz transmission started to happily accept the values being sent. So the transmitter was transmitting all along. Hard to see without a oscilloscope. Anyway, thank you for the responses and keep up the good work.

closed 2021-07-22 by Raynman77