LSatan/SmartRC-CC1101-Driver-Lib #84: Double CC1101 example needed

0x5c4r3, 2021-08-10:
There is no discussion, so I guess this is the best way to contact you.

Since you added new features to deal with two cc1101, it would be nice to see an example using such functions (two CC1101 at the same time in 1 board, maybe one receiving and one sending?).
It's not really clear how to use functions like addGDO0() or addGDO().

Thanks

MCKLtech, 2022-01-24:
Did you ever get this working? I'm also curious if it's even possible with a ESP8266 / D1 Mini

LSatan, 2022-01-25:
Hi, I will see that I add an example with the next update. this function has been added for easier switching between multiple modules. currently this is used in the firmware of ecil crow rf. https://github.com/joelsernamoreno/EvilCrow-RF
Regards