LSatan/SmartRC-CC1101-Driver-Lib #85: ELECHOUSE_CC1101::GDO_Set not correct

asafteirobert, 2021-08-14:
Calling `ELECHOUSE_CC1101::setGDO(byte gdo0, byte gdo2)` causes the library to not work correctly.
This is because it calls `ELECHOUSE_CC1101::GDO_Set` which sets `pinMode(GDO0, OUTPUT);`. It should be pinMode(GDO0, INPUT);

LSatan, 2021-08-14:
well I tested everything again and it worked perfectly right away. Make sure you are using setGdo0 and not set gdo. as it is written in the new examples. You can ignore the normal set Gdo function. setGdo0 is all about Input! Take a look at the cpp. Regards

LSatan, 2021-08-14:
[image]

asafteirobert, 2021-08-14:
Then what is the point of `setGDO` and `GDO_Set` methods? If they should not be used then they should not exist.

LSatan, 2021-08-14:
they use setGdo0 (your gdo0 pin); this is external. the blibliotehk uses gdo0set internally. It is used, but not controlled by the user. setgdo (your gdo0 pin, your gdo2 pin); is intended for bibliotehken who use the pins serial or analog. therefore gdo0 is an output tx and gdo2 is an input rx. so the old method cannot be used for the internal transfer functions. that's why there is now set gdo0. since gdo2 is not needed for this. if you do not know the difference between the internal (data sent digitally to the cc1101) and the external (anlog data sent to the cc1101) then read this explanation https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/31#issuecomment-624279213 . so if you only want to use the internal functions. then do it as in the examples.