LSatan/SmartRC-CC1101-Driver-Lib #86: CC1101 does not receive or transmit anything

VIKVIXX, 2021-08-20:
I bought TI-CC1101 and used it with arduino uno to sniff but when i uploaded any code from SmartRC library only thing i get on serial monitor is "Connection OK" i tried already like 6 RF remotes with 433.92 MHz but nothing happened so i bought same but new CC1101 module i also bought arduino mega and logical converter and after was everything was connected and i upload code it said "connection error" but without converter it said again "Connection OK" but still doesnt recieve or transmit anything. what should i do?

0x5c4r3, 2021-08-21:
So, in my experience, I am using an ESP32 with a CC1101. I am able to transmit without problems, but when I try to receive a signal sent from a cheap Radio doorbell transmitting at 433.92 MHz, nothing is detected. Don't know if it's a protocol problem or anything else.

VIKVIXX, 2021-08-21:
I tried using my garage door remote
Other garage remote
house alarm remote
And relay RF remote but not a single one was captured

zillawilli, 2022-01-16:
Hello
I have the same configuration as you (Nano and CC1101) and the same problem.

`#endif 

    Serial.begin(9600);
    
    if (ELECHOUSE_cc1101.getCC1101()){         // Check the CC1101 Spi connection.
    Serial.println("Connection OK");
    **Serial.println(gdo0);**
    }else{
    Serial.println("Connection Error");
    }
  
    ELECHOUSE_cc1101.Init();               // must be set to initialize the cc1101!
    ELECHOUSE_cc1101.setGDO0(gdo0);         // set lib internal gdo pin (gdo0). Gdo2 not use for this example.
   // ELECHOUSE_cc1101.setCCMode(1);       // set config for internal transmission mode.
    ELECHOUSE_cc1101.setModulation(2);  // set modulation mode. 0 = 2-FSK, 1 = GFSK, 2 = ASK/OOK, 3 = 4-FSK, 4 = MSK.
    ELECHOUSE_cc1101.setMHZ(433.92);   // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
    ELECHOUSE_cc1101.setSyncMode(2);  // Combined sync-word qualifier mode. 0 = No preamble/sync. 1 = 16 sync word bits detected. 2 = 16/16 sync word bits detected. 3 = 30/32 sync word bits detected. 4 = No preamble/sync, carrier-sense above threshold. 5 = 15/16 + carrier-sense above threshold. 6 = 16/16 + carrier-sense above threshold. 7 = 30/32 + carrier-sense above threshold.
 // ELECHOUSE_cc1101.setPA(10);      // set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12) Default is max!
    ELECHOUSE_cc1101.setCrc(1);     // 1 = CRC calculation in TX and CRC check in RX enabled. 0 = CRC disabled for TX and RX.
    **Serial.println("Tx Mode");**
}

void loop() {
`
It was not able to read the line "Tx Mode" in the serial monitor.
Best regards

frspp, 2023-04-06:
Same here - does not receive anything. 

Nodemcu ESP32 (ESP32-D0WDQ6) https://www.berrybase.de/esp32-nodemcu-development-board
TI CC1101 https://www.makershop.de/module/funk/cc1101-433mhz/

esrsound, 2024-02-28:
Same here, esp32 s3.