LSatan/SmartRC-CC1101-Driver-Lib #93: Please add config value for Oscillator Frequency

Gemikro1, 2021-11-07:
Not all of the CC1101 modules are using the default 26 MHz crystal as e.g. the Anaren A1101R08 series which is using a 27MHz crystal.
Could you please add an option to define the crystal frequency and use this value throughout the library to calculate the necessary frequencies and timings as in the functions .setMHZ() /.setDeviation() / .setChsp() etc. ?

LSatan, 2022-01-25:
Hi, Apologize for the late message. I will add it in the next update.

Gemikro1, 2022-01-25:
Perfect, thanks.
If you need someone for testing and feedback just tell me. 
Unfortunatly the Anaren devices are out of stock till somewhen in 2023, i realy liked them for building small 868MHz gateways :(