LSatan/SmartRC-CC1101-Driver-Lib #98: new receive simple method crashes

boerti71, 2022-01-19:
Hi

I get the following crash report:
Connection OK
Rx Mode
Rssi: -92
LQI: 128
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40085eba  PS      : 0x00060730  A0      : 0x800d2394  A1      : 0x3ffb1ec0  
A2      : 0x00000200  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000006  A8      : 0x80085bf8  A9      : 0x3ffb1ea0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060720  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000240  LBEG    : 0x400d233c  LEND    : 0x400d2349  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40085eba:0x3ffb1ec0 0x400d2391:0x3ffb1f00 0x400d1c7c:0x3ffb1f20 0x400d102d:0x3ffb1f40 0x400d1ac2:0x3ffb1f60 0x400d0da7:0x3ffb1f90 0x400d2d21:0x3ffb1fb0 0x40086189:0x3ffb1fd0

Rebooting...

I'm using ESP32_DEVKIT-V4 and [CC1101 868MHz module (green one)](https://www.aliexpress.com/item/1005002603812116.html?spm=a2g0o.9042311.0.0.29474c4dcp78fG)
header configuration is:

 ELECHOUSE_cc1101.Init();              // must be set to initialize the cc1101!
    ELECHOUSE_cc1101.setCCMode(1);       // set config for internal transmission mode.
    ELECHOUSE_cc1101.setModulation(0);  // set modulation mode. 0 = 2-FSK, 1 = GFSK, 2 = ASK/OOK, 3 = 4-FSK, 4 = MSK.
    ELECHOUSE_cc1101.setMHZ(868);   // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
    ELECHOUSE_cc1101.setSyncMode(0);  // Combined sync-word qualifier mode. 0 = No preamble/sync. 1 = 16 sync word bits detected. 2 = 16/16 sync word bits detected. 3 = 30/32 sync word bits detected. 4 = No preamble/sync, carrier-sense above threshold. 5 = 15/16 + carrier-sense above threshold. 6 = 16/16 + carrier-sense above threshold. 7 = 30/32 + carrier-sense above threshold.
    ELECHOUSE_cc1101.setCrc(0);      // 1 = CRC calculation in TX and CRC check in RX enabled. 0 = CRC disabled for TX and RX.

any ideas?
Thanks

boerti71, 2022-01-21:
found problem... byte buffer[99] was to small, changed to buffer[999] and now it's working

closed 2022-01-21 by boerti71

jayateertha043, 2023-06-18:
@LSatan Why is the buffer more than 61 needed ?
Hope CC1101 needs only 61 bytes ?